PhotonVision/photonvision #1420: PhotonVision for Orange Pi 5 Max

Plugner, 2024-09-14:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

There isn't any image for the OPi 5 Max at the moment, the current way of running photonvision now is getting a retail image and running the .jar in it. 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

When a new github release happens, a OrangePi 5 Max image should also be given.  

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Running photonvision JAR file inside another distro.

mcm001, 2024-09-14:
Just one more orange pi variant bro it'll be the last we swear just one more RK3588 board

More seriously: this would be done by making a new image be generated by photon-image-modifier and pulling it in. Look at how we added the 5+ for reference on how you can do this.

Plugner, 2024-09-16:
> Just one more orange pi variant bro it'll be the last we swear just one more RK3588 board
> 
> More seriously: this would be done by making a new image be generated by photon-image-modifier and pulling it in. Look at how we added the 5+ for reference on how you can do this.

Done! [https://github.com/PhotonVision/photon-image-modifier/pull/29](Pull Request)

mcm001, 2024-09-16:
Merged!

crschardt, 2024-09-17:
Thank you for adding the OPi5 Max to the images available. I've made a new release [v2025.0.0-beta-5](https://github.com/PhotonVision/photon-image-modifier/releases/tag/v2025.0.0-beta-5) of the image modifier that will include the OPi5 Max image. We also need a PR against the photonvision repository to add the OPi5 Max to the images generated when a new version of photonvision is released.

Some testing of the image would be nice too :)

crschardt, 2024-09-17:
There's a problem with the release on that tag. I'm working on a fix now. Until I have the fix, there won't be an artifact for you to reference. I'll let you know when I the fix is in.

LTCjRet, 2024-09-17:
@Plugner, you indicated that it might be possible to run a JAR file inside another opi5max distribution. Has this been tried?

crschardt, 2024-09-17:
> @Plugner, you indicated that it might be possible to run a JAR file inside another opi5max distribution. Has this been tried?

That's exactly what we do with the images. You can manually install photonvision on your own distro by following the instructions for [Other Debian-Based Co-Processor Installation](https://docs.photonvision.org/en/latest/docs/installation/sw_install/other-coprocessors.html
) in the documentation.

There are also some testable OPi5Max image [ready on my fork](https://github.com/crschardt/photon-image-modifier/releases/tag/v2025.0.0-alpha-3), and it should show up within an hour on the photon-image-modifier releases page.

LTCjRet, 2024-09-18:
@crschardt, I was able to get your image https://github.com/crschardt/photon-image-modifier/releases/tag/v2025.0.0-alpha-3 up and running on the Orange Pi 5 Max with minimal issues (purely due to my inexperience).  It runs with no obvious problems within the scope of basic calibration and AprilTag detection. Hardware links: opi5max - https://a.co/d/8IzH1JS and Arducam OV9782 - https://a.co/d/aQPdVmT If I can be of any assistance in testing or anything else you can think of I'd be more than happy to help.  Thanks so much!

LTCjRet, 2024-09-18:
@Plugner, please see my response to Craig above. I was able to flash and run his image on an opi5max 16Gb board.

crschardt, 2024-09-19:
@Plugner - the OPi5max image is available at https://github.com/PhotonVision/photon-image-modifier/releases/tag/v2025.0.0-beta-6

This image includes Photonvision v2024.3.1. If you want to test a newer version, you can download the `photonvision-dev-*-linuxarm64.jar` version from the Dev release and use the update button in the UI to upload it.

Now that the image seems to be working, the next step is to add the image to the `build-image` action in the photonvision workflow.

tyrius02, 2024-09-23:
How hard is it to install the image to eMMC?

crschardt, 2024-09-23:
I don’t know, but I really think that there’s no benefit. Photonvision boots in ~10 seconds on an OrangePi from uSD and its performance isn’t disk limited.

tyrius02, 2024-09-23:
See #1432.

crschardt, 2024-09-24:
I don’t see how that requires installing the image on the eMMC. Also, let’s move this discussion to that issue, since this is about making the image.

Juniormunk, 2024-10-25:
Fixed by #1444