PhotonVision/photonvision #1422: Changing to Aruco Pipeline crashes PV

asid61, 2024-09-15:
**Describe the bug**
Changing pipeline from Apriltag to Aruco on an OV9281 crashes PV until restarting it. After restart, the pipeline is restored to Apriltag.

**To Reproduce**
Steps to reproduce the behavior:
1. Make an Apriltag pipeline
2. Change it so an Aruco pipeline

**Screenshots / Videos**
I'll add these later

**Platform:**
 - Hardware Platform (ex. Raspberry Pi 4, Windows x64): Orange Pi 5
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch): VH-109
 - PhotonVision Version: photonvision-v2024.3.1
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Chrome
 - Camera(s) Used: OV9281, OV9782, Lifecam

**Additional context**
Add any other context about the problem here.

mcm001, 2024-09-15:
When you get a chance, journalCTL logs would also be helpful. That's the export current run logs button.

Juniormunk, 2024-09-15:
[photonvision-journalctl (1).txt](https://github.com/user-attachments/files/17004765/photonvision-journalctl.1.txt)

cb144p, 2024-09-15:
Might be linked to issue #1338

Juniormunk, 2024-10-06:
[hs_err_pid2383.log](https://github.com/user-attachments/files/17271125/hs_err_pid2383.log)
More info

mcm001, 2024-10-07:
Good find! A kaboom inside the wpilib apriltag detector is kinda terrifying.

closed 2024-10-09 by Juniormunk