PhotonVision/photonvision #1432: Save camera data to local storage

tyrius02, 2024-09-23:
**Is your feature request related to a problem? Please describe.**
As machine learning becomes more prevalent it is often useful to see what the camera and ML models are seeing to diagnose incorrect predictions. Today my only solution is to capture images. It would be useful, instead, if I could stream the video data into an mp4 file and store it on the device.

**Describe the solution you'd like**
Ideally, on the settings page of the UI there's be a toggle switch (record video ON/OFF) and a field where I can specify the video(s) should be stored.
When my robot enters auto or teleop the image data being read from the camera(s) would be streamed into an MP4 file.
An anonymous CIFS (samba) share could be added so we can easily download the data from the co-processor once a match has finished.

**Describe alternatives you've considered**
My driver station might be able to read the video stream from PhotonVision, but this just adds a bunch of data to the network and potentially causes other issues on the driver station.
Other alternatives? IDK.

**Additional context**
Only real major concern I'd have is storage space. I don't think it is wise to attempt to write video data to a tmpfs nor do I think it is wise to write it to the SD card. Perhaps we can plug a USB flash drive into the SBC and offer that as one solution. An M.2 NVME might be an option, eMMC, etc. It might be good to keep track of how much free space is left on the storage device and stop writing if we exceed something like 95% capacity.

Alextopher, 2024-09-29:
Related:  #766

mcm001, 2024-09-29:
I would argue probably duplicate of #766?

tyrius02, 2024-09-29:
Yup

tyrius02, 2024-09-29:
Dup of #766

closed 2024-09-29 by tyrius02