PhotonVision/photonvision #1452: Unable to Load PhotonVision On Romi

EmDash00, 2024-10-10:
Following the instructions on the PhotonVision docs I installed PhotonVision using the install script for my Romi; however, PhotonVision appears to crash upon starting up.

Here is the relevant section of the journalctl logs below. I've tried this on the 2023.2.1 release (the latest stable release). Oddly enough also get GLIBCXX errors since it requests a newer version than is available on that image. Here's the relevant section from the journalctl logs for your reference. The top of the log is just the initial bootup of PhotonVision; however, I set the romi to read/write mode using its settings client and restarted PhotonVision to see if it'd make a difference (unfortunately no).

```
Oct 09 19:20:30 photonvision java[1063]: java.lang.UnsatisfiedLinkError: /tmp/libphotonlibcamera.so: libcamera.so.0.1: cannot open shared object file: No such file or directory
Oct 09 19:20:30 photonvision java[1063]:         at java.base/jdk.internal.loader.NativeLibraries.load(Native Method)
Oct 09 19:20:30 photonvision java[1063]:         at java.base/jdk.internal.loader.NativeLibraries$NativeLibraryImpl.open(NativeLibraries.java:388)
Oct 09 19:20:30 photonvision java[1063]:         at java.base/jdk.internal.loader.NativeLibraries.loadLibrary(NativeLibraries.java:232)
Oct 09 19:20:30 photonvision java[1063]:         at java.base/jdk.internal.loader.NativeLibraries.loadLibrary(NativeLibraries.java:174)
Oct 09 19:20:30 photonvision java[1063]:         at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2389)
Oct 09 19:20:30 photonvision java[1063]:         at java.base/java.lang.Runtime.load0(Runtime.java:755)
Oct 09 19:20:30 photonvision java[1063]:         at java.base/java.lang.System.load(System.java:1953)
Oct 09 19:20:30 photonvision java[1063]:         at org.photonvision.raspi.LibCameraJNILoader.forceLoad(LibCameraJNILoader.java:64)
Oct 09 19:20:30 photonvision java[1063]:         at org.photonvision.Main.main(Main.java:365)
Oct 09 19:20:30 photonvision java[1063]: [2024-10-09 19:20:30] [General - Main] [ERROR] Platform does not support RKNN based machine learning!
Oct 09 19:20:30 photonvision java[1063]: [2024-10-09 19:20:30] [Camera - MrCalJNILoader] [ERROR] Couldn't load shared object mrcal_jni: /tmp/libmrcal_jni.so: /lib/aarch64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /tmp/libmrcal_jni.so)
Oct 09 19:20:30 photonvision java[1063]: java.lang.UnsatisfiedLinkError: /tmp/libmrcal_jni.so: /lib/aarch64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /tmp/libmrcal_jni.so)
Oct 09 19:20:30 photonvision java[1063]:         at java.base/jdk.internal.loader.NativeLibraries.load(Native Method)
Oct 09 19:20:30 photonvision java[1063]:         at java.base/jdk.internal.loader.NativeLibraries$NativeLibraryImpl.open(NativeLibraries.java:388)
Oct 09 19:20:30 photonvision java[1063]:         at java.base/jdk.internal.loader.NativeLibraries.loadLibrary(NativeLibraries.java:232)
Oct 09 19:20:30 photonvision java[1063]:         at java.base/jdk.internal.loader.NativeLibraries.loadLibrary(NativeLibraries.java:174)
Oct 09 19:20:30 photonvision java[1063]:         at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2389)
Oct 09 19:20:30 photonvision java[1063]:         at java.base/java.lang.Runtime.load0(Runtime.java:755)
Oct 09 19:20:30 photonvision java[1063]:         at java.base/java.lang.System.load(System.java:1953)
Oct 09 19:20:30 photonvision java[1063]:         at org.photonvision.jni.PhotonJNICommon.forceLoad(PhotonJNICommon.java:64)
Oct 09 19:20:30 photonvision java[1063]:         at org.photonvision.mrcal.MrCalJNILoader.forceLoad(MrCalJNILoader.java:65)
Oct 09 19:20:30 photonvision java[1063]:         at org.photonvision.Main.main(Main.java:380)
```

Also appears to successfully fetch the temperature and other statistics, but I don't see them on the client.

```
Oct 09 19:23:49 photonvision java[1063]: [2024-10-09 19:23:49] [General - MetricsManager] [DEBUG] Publishing Metrics...
Oct 09 19:23:49 photonvision java[1063]: [2024-10-09 19:23:49] [General - ShellExec] [DEBUG] Executing "sed 's/.\{3\}$/.&/' /sys/class/thermal/thermal_zone0/temp"
Oct 09 19:23:49 photonvision java[1063]: [2024-10-09 19:23:49] [General - ShellExec] [DEBUG] Got exit code 0
Oct 09 19:23:49 photonvision java[1063]: [2024-10-09 19:23:49] [General - ShellExec] [DEBUG] Executing "top -bn1 | grep "Cpu(s)" | sed "s/.*, *\([0-9.]*\)%* id.*/\1/" | awk '{print 100 - $1}'"
Oct 09 19:23:50 photonvision java[1063]: [2024-10-09 19:23:50] [General - ShellExec] [DEBUG] Got exit code 0
Oct 09 19:23:50 photonvision java[1063]: [2024-10-09 19:23:50] [General - ShellExec] [DEBUG] Executing "free -m | awk 'FNR == 2 {print $2}'"
Oct 09 19:23:50 photonvision java[1063]: [2024-10-09 19:23:50] [General - ShellExec] [DEBUG] Got exit code 0
Oct 09 19:23:50 photonvision java[1063]: [2024-10-09 19:23:50] [General - ShellExec] [DEBUG] Executing "if   ((  $(( $(vcgencmd get_throttled | grep -Eo 0x[0-9a-fA-F]*) & 0x01 )) != 0x00 )); then echo "LOW VOLTAGE";  elif ((  $(( $(vcgencmd get_throttled | grep -Eo 0x[0-9a-fA-F]*) & 0x08 )) != 0x00 )); then echo "HIGH TEMP";  elif ((  $(( $(vcgencmd get_throttled | grep -Eo 0x[0-9a-fA-F]*) & 0x10000 )) != 0x00 )); then echo "Prev. Low Voltage";  elif ((  $(( $(vcgencmd get_throttled | grep -Eo 0x[0-9a-fA-F]*) & 0x80000 )) != 0x00 )); then echo "Prev. High Temp";  else echo "None"; fi"
Oct 09 19:23:50 photonvision java[1063]: [2024-10-09 19:23:50] [General - ShellExec] [DEBUG] Got exit code 0
Oct 09 19:23:50 photonvision java[1063]: [2024-10-09 19:23:50] [General - ShellExec] [DEBUG] Executing "uptime -p | cut -c 4-"
Oct 09 19:23:50 photonvision java[1063]: [2024-10-09 19:23:50] [General - ShellExec] [DEBUG] Got exit code 0
Oct 09 19:23:50 photonvision java[1063]: [2024-10-09 19:23:50] [General - ShellExec] [DEBUG] Executing "vcgencmd get_mem gpu | grep -Eo '[0-9]+'"
Oct 09 19:23:50 photonvision java[1063]: [2024-10-09 19:23:50] [General - ShellExec] [DEBUG] Got exit code 0
Oct 09 19:23:50 photonvision java[1063]: [2024-10-09 19:23:50] [General - ShellExec] [DEBUG] Executing "free -m | awk 'FNR == 2 {print $3}'"
Oct 09 19:23:50 photonvision java[1063]: [2024-10-09 19:23:50] [General - ShellExec] [DEBUG] Got exit code 0
Oct 09 19:23:50 photonvision java[1063]: [2024-10-09 19:23:50] [General - ShellExec] [DEBUG] Executing "vcgencmd get_mem malloc | grep -Eo '[0-9]+'"
Oct 09 19:23:50 photonvision java[1063]: [2024-10-09 19:23:50] [General - ShellExec] [DEBUG] Got exit code 0
Oct 09 19:23:50 photonvision java[1063]: [2024-10-09 19:23:50] [General - ShellExec] [DEBUG] Executing "df ./ --output=pcent | tail -n +2"
Oct 09 19:23:50 photonvision java[1063]: [2024-10-09 19:23:50] [General - ShellExec] [DEBUG] Got exit code 0
Oct 09 19:23:50 photonvision java[1063]: [2024-10-09 19:23:50] [WebServer - Server] [DEBUG] Handled HTTP request of type POST from endpoint /api/utils/publishMetrics for host 10.42.0.1 in 474.6116 ms
Oct 09 19:23:57 photonvision java[1063]: [2024-10-09 19:23:57] [General - ShellExec] [DEBUG] Executing "journalctl -u photonvision.service > /tmp/photonvision-journalctl592160893```

**Screenshots / Videos**

![image](https://github.com/user-attachments/assets/1312ff3e-094f-4aed-abe5-83a8f342da9f)

**Platform:**
 - Raspberry Pi 3 Raspbian 64-bit
 - Connected over ethernet to my laptop (LAN only)
 - PhotonVision Version: v2024.3.1 
 - Browser (with Version): Firefox 131.0 (64-bit)

Juniormunk, 2024-10-10:
it appears that you are missing the libcamera dependency for that version.  Im not sure what version of raspbian you are using but you could try ```sudo apt-get install libcamera0.1 ```

mcm001, 2024-10-10:
That's part of it. But the base rock image is just too old to be supported. Unfortunately not sure if you can simply install a newer glibc -- maybe some people have tried docker? But effectively the romi isn't really supported until they update their base image. That's a wpilib limitation, not just us

EmDash00, 2024-10-10:
Installing a new glibc would be impossible unless I wanted to build it from source I assume. Is there a version that worked in the past? I might just load an older image.

mcm001, 2024-10-10:
I believe 2023 should have the proper glibc dependency for you? But you'll have to recompile photonlib yourself, as I'm not sure if we still have those artifacts live. Might not be that bad just something to be aware of.

mcm001, 2024-10-10:
In any case we need to update our romi docs to reflect this all

Juniormunk, 2024-10-10:
I think that this might still be a bug in the latest dev. If libcamera can't be found it just shuts the whole thing down.

EmDash00, 2024-10-10:
Well I'm mostly just trying to find a way as a mentor to help kids practice without a full robot. I was hoping they could use PhotonVision with a usb camera to try aligning to some target with a Romi. If you have any advice on how I should proceed, that'd be good, in particular, which version of PhotonVision to build for which version of WPILibPi? From what I saw earlier, I should try a 2023 version of PhotonLib which should target the 2023 releases of WPILibPi? I assume I can just follow the manual install tips.

EmDash00, 2024-10-10:
> I believe 2023 should have the proper glibc dependency for you? But you'll have to recompile photonlib yourself, as I'm not sure if we still have those artifacts live. Might not be that bad just something to be aware of.

So my solution ended up being downloading the v2023.4.2 version of PhotonVision from the releases page. I think it might be good to be able to specify which version of PhotonVision is downloaded by the installer script; however, we might need more useful error information displayed for version mismatch if it's possible. This took way too long to debug otherwise. I might be able to work on updating the docs in my freetime.

EmDash00, 2024-10-11:
@mcm001 I've made a patch to the install script that makes this workaround easier (and other improvements). See PR https://github.com/PhotonVision/photonvision/pull/1456. Let me now your thoughts.