PhotonVision/photonvision #1491: Add a stable version of the PhotonVision Read the Docs. It currently only has the latest dev version.

VStellar2023, 2024-10-26:
**Is your feature request related to a problem? Please describe.**

The code in the "Aiming at a Target" example java code has a method called getAllUnreadResults(), which does not exist in the PhotonLib library. I believe this is because the PhotonVision Read the docs only have a latest development version published (which I think is not fully updated/debugged?); they do not have a stable version published.

**Describe the solution you'd like**

A stable version of the PhotonVision Read the Docs should be published.

**Describe alternatives you've considered**

N/A - I fixed this problem by just grabbing the PhotonLib zip file from the latest development version in GitHub, which included this method. After that the issue was solved.

**Additional context**

It was giving me the error: " the method getAllUnreadResults() is undefined for the type PhotonCamera."

mcm001, 2024-10-26:
Looks easy to do from the read the docs side https://docs.readthedocs.io/en/stable/versions.html

mcm001, 2024-11-02:
docs.photonvision.org now defaults to https://docs.photonvision.org/en/v2025.0.0-alpha-0/ and has an option for latest if you want it. this is 90% of the way there, api docs are still unversioned.

mcm001, 2024-11-02:
API docs covered by #1378

Sam948-byte, 2025-01-08:
This looks like it was finished, and can probably be closed.

closed 2025-02-02 by VStellar2023