PhotonVision/photonvision #1507: Switch protobuf Java to generating with a Python script

spacey-sooty, 2024-10-29:


mcm001, 2024-10-29:
Why? Keeping things in gradle makes tracking task inputs and outputs easier.

spacey-sooty, 2024-10-29:
> and is in line with wpilib.

WPILib doesn't do this in Gradle? https://github.com/wpilibsuite/allwpilib/blob/f03e0cdf6a86e8fdb4111bf76d64f73539f19936/wpimath/generate_quickbuf.py#L13

> Why? Keeping things in gradle makes tracking task inputs and outputs easier,

If we do keep it in Gradle we shouldn't use the plugin we are right now, we don't use its Java output anyways and we wont be using it for C++ after https://github.com/PhotonVision/photonvision/pull/1484

mcm001, 2024-10-29:
Oh since December 2023, TIL. do you know why they switched to committing the generated code?
(Committing generated code is based af, ftr)

spacey-sooty, 2024-10-29:
Something about devs being able to read it in the codebase or something

Gold856, 2024-10-29:
That's the reason for the other generated files, but the quickbuf files have a different reason. Having the quickbuf Java files pregenerated and committed means you don't need the quickbuf plugin available to generate the source files. This only impacts CMake because Gradle will download the plugin for you, and for Photon, there isn't any particular benefit to pregenerating quickbuf files (since CMake is not a build option) other than it's committed to the repo and you can look at it.

closed 2024-12-21 by spacey-sooty