PhotonVision/photonvision #1515: Strongly recommend charuco

Juniormunk, 2024-11-02:
There is a bug with our 8x8 chessboard that can result in bad calibration results with a small number of similar images. We should add a UI message that strongly recommends the charuco.

mdurrani808, 2024-11-05:
Is this a bug? Or is this not just a inherent property of calibration? If i take a small number of similar images of charuco, will this not happen?

mcm001, 2024-11-05:
It's inherent to mrcal+the opencv chessboard solver it seems? I'll dig up the old issue we had last year but it's specifically square chessboards

samfreund, 2025-04-14:
What exactly is the resolution here? Recommend users take lots of calibration images?

samfreund, 2025-04-14:
Solution is discourage chessboards.

mcm001, 2025-10-25:
OG reason that we should discourage chessboards was documented by Dima in https://github.com/PhotonVision/photonvision/issues/1120#issuecomment-1974330587