PhotonVision/photonvision #1612: obs object in calibrationUtils.py throws an incorrect class type

elvizer, 2024-11-30:
**Describe the bug**
When running calibrationUtils.py found in the latest version of the docs, I get this error:
`AttributeError: 'Generic' object has no attribute 'snapshotData'. Did you mean: 'snapshotName'?`

However, when printing the type of `obs` in `obs.snapshotData.data`, it prints:
`<class '__main__.convert_photon_to_mrcal.<locals>.Generic'>`

When hovering over `obs` in Visual Studios Code, it says that it is of type `Observation`

**To Reproduce**
Steps to reproduce the behavior:
1. Download calibrationUtils.py from the latest version of PhotonVision docs, found [here.](https://raw.githubusercontent.com/PhotonVision/photonvision/master/devTools/calibrationUtils.py)
2. Run `python3 /path/to/calibrationUtils.py path/to/photon_calibration.json /path/to/output/folder` (in my case WSL).
3. See error

**Screenshots / Videos**
[image]
[image]

**Platform:**
 - Windows 10 (Version 19045.5131)
 - Ubuntu 24.04.1 LTS (WSL)
 - Visual Studios Code

mcm001, 2024-12-07:
@Juniormunk did cal json format change over the summer? Either way I have a patched version floating around.

Juniormunk, 2024-12-07:
Yes. We no longer store the images in the json.

mcm001, 2024-12-12:
Maybe  https://github.com/PhotonVision/photonvision/pull/1615

PureRandomGit, 2024-12-13:
Sadly I still get the same error as above.
```
Traceback (most recent call last):
  File "/calibrationUtils.py", line 254, in <module>
    main()
  File "/calibrationUtils.py", line 250, in main
    convert_photon_to_mrcal(args.input, args.output_folder)
  File "/calibrationUtils.py", line 210, in convert_photon_to_mrcal
    image = obs.snapshotData.data
            ^^^^^^^^^^^^^^^^
AttributeError: 'Generic' object has no attribute 'snapshotData'. Did you mean: 'snapshotName'?
```
Platform:
KDE Neon6.2
Photonvision v2025.0.0-beta-6

Juniormunk, 2024-12-13:
@PureRandomGit  Sorry. I meant to fix this a while ago. I don't have a good way to test this code, but here are the changes. #1633

PureRandomGit, 2024-12-14:
Thank you, it fixes that problem but I encountered two others so far.
First is this one which I bypassed by changing the json width to remove the floats
```
Traceback (most recent call last):
  File "/calibrationUtils.py", line 237, in <module>
    main()
  File "/calibrationUtils.py", line 233, in main
    convert_photon_to_mrcal(args.input, args.output_folder)
  File "/calibrationUtils.py", line 210, in convert_photon_to_mrcal
    mrcal_model = __convert_cal_to_mrcal_cameramodel(camera_cal_data)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/calibrationUtils.py", line 129, in __convert_cal_to_mrcal_cameramodel
    ).reshape((cal.calobjectSize.width, cal.calobjectSize.height, 3))
      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: 'float' object cannot be interpreted as an integer
```
At this point it was able to generate the corners.vnl but errors when trying to reshape the array due to some of the images not getting all the points, at least I think.
```
Traceback (most recent call last):
  File "/calibrationUtils.py", line 237, in <module>
    main()
  File "/calibrationUtils.py", line 233, in main
    convert_photon_to_mrcal(args.input, args.output_folder)
  File "/calibrationUtils.py", line 210, in convert_photon_to_mrcal
    mrcal_model = __convert_cal_to_mrcal_cameramodel(camera_cal_data)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/calibrationUtils.py", line 129, in __convert_cal_to_mrcal_cameramodel
    ).reshape((cal.calobjectSize.width, cal.calobjectSize.height, 3))
      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: cannot reshape array of size 135 into shape (7,7,3)
```

PR #2055 (2025-08-13): Fix calibrationUtils.py
## Description

This brings the representation of calibration data in calibrationUtils.py up-to-date with what is in photon-core. This includes Size actually containing floats, not ints, and the removal of snapshotData. This also fills in missing corners with negatively weighted points so that it will match the new array shape without impacting the calibration results.

Closes #1612, closes #1615, closes #1633, closes #1656.

## Meta

Merge checklist:
- [x] Pull Request title is [short, imperative summary](https://cbea.ms/git-commit/) of proposed changes
- [x] The descr [... cut]