PhotonVision/photonvision #1626: Expose WPLIB switched camera to allow quick driver station switching between cameras

jsimpso81, 2024-12-06:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Create a "virtual" switched camera that allows a single stream to the dashboard to switch between cameras by writing network table value to the dashboard.  This functionality exists on the WPILIBPI.   

**Describe alternatives you've considered**
None that are really practical.

**Additional context**
None

mcm001, 2024-12-07:
Wdym switched camer?

jsimpso81, 2024-12-07:
Wpilibpi has a virtual camera that can stream back to the dashboard.   The virtual camera can instantly switch its input to a different physical camera by changing the value of a network table variable written to the coprocessor.    This happens quickly and doesn’t disrupt the dashboard stream.   It is useful to give drivers different views ( intake, vs field view, for example ).

mcm001, 2024-12-07:
That feels like a feature that belongs as part of the dashboard, not photon, to me?

jsimpso81, 2024-12-07:
It takes time to switch h to a new stream from the dashboard even if it is programmatically controlled.   ( new socket connection and other negotiation, having the  coprocessor switch makes it work faster and smoother. ( hence the reason it was in wpilibpi ).

jsimpso81, 2025-01-20:
I assumed that this would be easy to do since it was already done in Wpilibpi.