PhotonVision/photonvision #1628: Limelight 2+ not booting with version 2025.0.0-beta-5

Astatin3, 2024-12-07:
**Describe the bug**
Using a limelight 2+, and 2025.0.0-beta-5 the device does not appear to boot.

**To Reproduce**
Steps to reproduce the behavior:
Flash the image with the recommended version of Balena Etcher

**Platform:**
 - Hardware Platform (ex. Raspberry Pi 4, Windows x64): Limelight 2+
 - Network Configuration: Laptop and limelight connected via router over ethernet.
 - PhotonVision Version: 2025.0.0-beta-5
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): 2025.0.0-beta-5
 - Camera(s) Used: Whatever the Limelight 2+ has.

**Additional context**
- I was not at all using the most ideal setup, with a linux laptop writing data, and an undervolted limelight.
- I redownloaded and reflashed the image in case it was download error
- Limelight version 2024.10.2 flashed and worked fine
- Photonvision version 2024.3.1 flashed and worked fine

Juniormunk, 2024-12-07:
Oh boy so much fun. Can you hook up a screen to it and send the console outputs?

Astatin3, 2024-12-07:
With a working photon release, I can't get an HDMI output working over the USB-A port on the device, Is there a way I could view the logs outputted to the local storage?

Astatin3, 2024-12-07:
After a fresh flash of 2024.3.1, I attempted to install beta-5 using the jar upload feature, the jar ran and the UI came up, but it didn't seem to detect any hardware. (Look at image below)
[image]

I can try some more monitors in a couple days when I have access to them.

Juniormunk, 2024-12-07:
Ahhh right I keep forgetting that limelight doesn't have a video output. Does it show up on the network? Can you ssh into it?

Astatin3, 2024-12-07:
I don't believe so, but give me a minute to reflash the image and see if I can just in case.

Astatin3, 2024-12-07:
Okay, for some reason it's working now. This is my first time working with PhotonVision, and I saw the green status LED that turns on with power inside the case of the limelight was blinking, which made me think it was some kind of bootloop. Thank you so much for your help! This was really just stupid on my end.

closed 2024-12-07 by Astatin3