PhotonVision/photonvision #1634: Incorrect pose estimation vision correction java example

PGgit08, 2024-12-15:
In the PhotonLib pose estimation java example, old vision measurements are being ignored and only the most recent measurement from `getAllUnreadResults()` is being added to the WPILib pose estimator:

- `getEstimatedGlobalPose()` method **(the issue is in this method)**: https://github.com/PhotonVision/photonvision/blob/e8efef476b3b4681c8899a8720774d6dbd5ccf56/photonlib-java-examples/poseest/src/main/java/frc/robot/Vision.java#L86-L114

- `SwerveDrivePoseEstimator` vision update call using the `getEstimatedGlobalPose()` method: https://github.com/PhotonVision/photonvision/blob/e8efef476b3b4681c8899a8720774d6dbd5ccf56/photonlib-java-examples/poseest/src/main/java/frc/robot/Robot.java#L64-L73

PGgit08, 2024-12-15:
i forgot to add the bug label

mcm001, 2024-12-15:
Yeah agreed that that example looks like an anti pattern. Good catch!

gerth2, 2024-12-21:
Yee. The example design predates `getAllUnreadResults()`, was just never updated.

gerth2, 2024-12-21:
Fyi some overlap with #1632

closed 2025-04-15 by samfreund