PhotonVision/photonvision #1635: OpenCV loaded failed with WPILib 2025.1.1-beta-3 when simulating

RockyXRQ, 2024-12-18:
**Describe the bug**
OpenCV loaded failed with WPILib 2025.1.1-beta-3 when simulating

**To Reproduce**
Steps to reproduce the behavior:
1. just write `var sim = new PhotonCameraSim(camera, simCameraProperties)` in code
2. run `Simulate Robot Code`

**Screenshots / Videos**
[image]

**Platform:**
 - Hardware Platform: Windows x64
 - Network Configuration: LocalHost
 - PhotonVision Version: v2025.0.0-beta-6
 - Browser: None
 - Camera(s) Used: None

mcm001, 2024-12-18:
Is opencv_java480d in the jar at the path it tried to extract from? Are you positive that you're using the jdk shipped with wpilib 2025 beta 2?

RockyXRQ, 2024-12-19:
Seems I don't have that jar at the path, only bunch of .dll and .pdb file. I can run the simulator in 2025 beta 2 but not after updated to 2025 beta 3

RockyXRQ, 2024-12-19:
I'm using a totally freshly installed wpilib 2025 vscode, so I guess I'm using the the jdk shipped with wpilib 2025 beta 2? Is there anyway to confrim that?

RockyXRQ, 2024-12-19:
I have these in build/jni/release
[image]

mcm001, 2024-12-19:
Huh. So we're trying to load 4.8, but we only ship 4.10?

spacey-sooty, 2024-12-19:
WPILib recently updated

RockyXRQ, 2024-12-19:
Seems they did update it to 4.10 https://github.com/wpilibsuite/allwpilib/commit/c3fc7c829d2251bb65f29cbed452df1e8883c07d

mcm001, 2024-12-19:
Wait hold on. Wat. You're using Beta 3? we only currently support beta 2. https://github.com/PhotonVision/photonvision/blob/master/build.gradle#L33

Try again using beta 2, or wait till we put out a release.

mcm001, 2024-12-19:
I can run the sim fine using 2025 wpilib beta 2 vscode still. Reopen if you have issues using beta 2 still!

closed 2024-12-19 by mcm001