PhotonVision/photonvision #1640: White Balance shown fopr known Black/white cameras

gerth2, 2024-12-19:
reproduced on beta6 on an opi 5

[image]

white balance input does nothing, shouldn't need to show it

mcm001, 2024-12-19:
Does V4L list white balance as a property, it just doesn't do anything?

gerth2, 2024-12-19:
```
[2024-12-19 17:27:26] [Camera - USBCameraSource - Arducam_OV2311_USB_Camera] [DEBUG] Cam Properties Dump:
Name: connect_verbose, Kind: kInteger, Value: 2, Min: 0, Max: 1, Dflt: 1, Step: 1
Name: , Kind: kNone, Value: 0, Min: 0, Max: 0, Dflt: 0, Step: 0
Name: raw_brightness, Kind: kInteger, Value: 2, Min: -64, Max: 64, Dflt: 0, Step: 1
Name: brightness, Kind: kInteger, Value: 2, Min: 0, Max: 100, Dflt: 50, Step: 1
Name: raw_contrast, Kind: kInteger, Value: 2, Min: 0, Max: 64, Dflt: 32, Step: 1
Name: contrast, Kind: kInteger, Value: 2, Min: 0, Max: 100, Dflt: 50, Step: 1
Name: raw_saturation, Kind: kInteger, Value: 2, Min: 0, Max: 128, Dflt: 64, Step: 1
Name: saturation, Kind: kInteger, Value: 2, Min: 0, Max: 100, Dflt: 50, Step: 1
Name: raw_hue, Kind: kInteger, Value: 2, Min: -40, Max: 40, Dflt: 0, Step: 1
Name: hue, Kind: kInteger, Value: 2, Min: 0, Max: 100, Dflt: 50, Step: 1
Name: white_balance_automatic, Kind: kBoolean, Value: 1, Min: 0, Max: 1, Dflt: 1, Step: 1
Name: gamma, Kind: kInteger, Value: 2, Min: 72, Max: 500, Dflt: 100, Step: 1
Name: raw_gain, Kind: kInteger, Value: 2, Min: 0, Max: 100, Dflt: 0, Step: 1
Name: gain, Kind: kInteger, Value: 2, Min: 0, Max: 100, Dflt: 0, Step: 1
Name: power_line_frequency, Kind: kEnum, Value: 8, Min: 0, Max: 2, Dflt: 2, Step: 1
Name: white_balance_temperature, Kind: kInteger, Value: 2, Min: 2800, Max: 6500, Dflt: 4600, Step: 1
Name: raw_sharpness, Kind: kInteger, Value: 2, Min: 0, Max: 6, Dflt: 3, Step: 1
Name: sharpness, Kind: kInteger, Value: 2, Min: 0, Max: 100, Dflt: 50, Step: 1
Name: backlight_compensation, Kind: kInteger, Value: 2, Min: 0, Max: 2, Dflt: 1, Step: 1
Name: auto_exposure, Kind: kEnum, Value: 8, Min: 0, Max: 3, Dflt: 3, Step: 1
Name: raw_exposure_time_absolute, Kind: kInteger, Value: 2, Min: 1, Max: 5000, Dflt: 157, Step: 1
Name: exposure_time_absolute, Kind: kInteger, Value: 2, Min: 0, Max: 100, Dflt: 3, Step: 1
Name: exposure_dynamic_framerate, Kind: kBoolean, Value: 1, Min: 0, Max: 1, Dflt: 0, Step: 1
```

[image]

gerth2, 2024-12-19:
well maybe leave it in for now. Expect user questions. And/or `IgnoreWhiteBalance` quirk.