PhotonVision/photonvision #1651: Docs: Update all 2024 references to 2025

gerth2, 2024-12-21:
Mostly related to wpilib releases, apriltag layouts, gamepiece detection.

Ctrl-f the repo for 2024, analyze each occurrence.

Sam948-byte, 2025-01-01:
I'd be willing to do this post-kickoff.  It seems like the docs repo has been archived, so if someone can point me to the current location of the repo, that would be helpful.

mcm001, 2025-01-01:
Docs now live in the source tree. I think that https://github.com/PhotonVision/photonvision/pull/1650 probably covers most of this?

Sam948-byte, 2025-01-01:
Gotcha, I think I found it.  I'll pop back prolly 1/6 and get this taken care of.  (I want to wait until after kickoff, since I'm sure there will be new documents released)

Sam948-byte, 2025-01-01:
Quick clarification, should I PR off #1650 and use that as my base?

mcm001, 2025-01-02:
It would probably be merged, but would love help updating any other links post-kickoff for sure!

PR #1685 (2025-01-07): Update references in docs to 2025
Fixes #1651
commit 36e3a1f7 (2025-01-07): Update references in docs to 2025 (#1685)

Fixes #1651
closed 2025-01-07 by mcm001