PhotonVision/photonvision #1656: ValueError when reshaping array with calibrationUtils.py

PureRandomGit, 2024-12-23:
I get this error when running the updated script for the mrcal [calibrationUtils.py](https://github.com/PhotonVision/photonvision/blob/dbfd3cccfd0d761077a3fd7d4d62d2b556036559/devTools/calibrationUtils.py) I get this error:
```
$ python3 calibrationUtils.py calibration2.json ./img
Traceback (most recent call last):
  File "calibrationUtils.py", line 237, in <module>
    main()
  File "calibrationUtils.py", line 233, in main
    convert_photon_to_mrcal(args.input, args.output_folder)
  File "calibrationUtils.py", line 210, in convert_photon_to_mrcal
    mrcal_model = __convert_cal_to_mrcal_cameramodel(camera_cal_data)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "calibrationUtils.py", line 129, in __convert_cal_to_mrcal_cameramodel
    ).reshape((cal.calobjectSize.width, cal.calobjectSize.height, 3))
      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: cannot reshape array of size 135 into shape (7,7,3)
```
I believe its caused by images that do not have all the points of the checkerboard being detected/in frame as when I run it with a calibration that has everything in frame and detected it appears to work fine and generates the cameramodel.

broken calibration file: 
[calibration2.json](https://github.com/user-attachments/files/18232963/calibration2.json)
working calibration file:
[calibration3.json](https://github.com/user-attachments/files/18233043/calibration3.json)

**Platform:**
 - Hardware Platform: Orange Pi 5
 - PhotonVision Version:  v2025.0.0-beta-8 
 - Camera(s) Used: OV9281

Juniormunk, 2025-01-02:
Is this using charuco calibration? I wonder if we are exporting all "fake" points as well.

PureRandomGit, 2025-01-02:
Yes, this is with charuco cal.

PR #2055 (2025-08-13): Fix calibrationUtils.py
## Description

This brings the representation of calibration data in calibrationUtils.py up-to-date with what is in photon-core. This includes Size actually containing floats, not ints, and the removal of snapshotData. This also fills in missing corners with negatively weighted points so that it will match the new array shape without impacting the calibration results.

Closes #1612, closes #1615, closes #1633, closes #1656.

## Meta

Merge checklist:
- [x] Pull Request title is [short, imperative summary](https://cbea.ms/git-commit/) of proposed changes
- [x] The descr [... cut]