PhotonVision/photonvision #1683: Additional PI Camera support

nlbutts, 2025-01-05:
**Is your feature request related to a problem? Please describe.**
I'm trying to add the V3, HQ, and V3 Wide camera support to PhotonVision. I'm a professional C/C++ and Python programmer, but my Java knowledge is pretty poor. How can I build **photon-libcamera-gl-driver** code and integrated it into the **PhotonVision** build?

If I understand things correctly, when **PhotonVision** builds, it pulls the **photon-libcamera-gl-driver** code from Maven. ChatGPT has failed me. It seems like I should be able to modify to a local repo, but I have yet to figure that out. So if anyone could point me in the right direction, I would be grateful. 

The second question is more of an architecture question. **PhotonVision** asks the **photon-libcamera-gl-driver** for a enum and friendly name for the camera. Why have to keep two Enum in sync in the two projects. Would it be better to integrate the sysfs path to camera name in **PhotonVision** and remove that from the **photon-libcamera-gl-driver**?

Thanks in Advance.

Juniormunk, 2025-01-05:
No code specifically blocks certain cameras all it does is give warnings that they are not supported. Except for specific camera features such as autofocus or setting the focus, those cameras should "work" (but are not supported) with photonvision.

mcm001, 2025-01-05:
(Although we do choose video modes based on friendly name)

The general workflow for using a local build of our libcamera driver (from memory, fact check me against our GitHub actions workflow):
- use a pi or chroot into a pi image 
- build with cmake
- use gradle to package the .so + Java code into a JAR (JARs? Not sure if Java and native code are seperate artifacts)
- you can use gradlew publishToMavenLocal to publish to a maven repository at ~/.m2/repository 
- update the libcamera driver version in photon. We should already be configured to search that local maven repository

nlbutts, 2025-01-06:
Any thoughts on the Pi AI camera. I have YOLOV8-Nano running on it at 30 FPS. Eventually I want to try and add support for that.