PhotonVision/photonvision #1697: Incompatible webcam?

brunsa2, 2025-01-09:
I bought an [inexpensive webcam](https://www.microcenter.com/product/684054/aluratek-hd-1080p-usb-webcam-w-manual-focus) to start testing with on an RPi 5. `v4l2-ctl --list-devices` and `v4l2-ctl --list-formats` both show the camera and a few video formats.

Excerpt of the logs:
```
[20:29:33] [2025-01-07 01:29:32] [WebServer - DataChangeService] [DEBUG] Added subscriber - Sources: DCS_WEBSOCKET, DCS_HTTP, DCS_NETWORKTABLES, DCS_VISIONMODULE, DCS_OTHER, Destinations: DCD_WEBSERVER
[20:29:33] [2025-01-07 01:29:32] [WebServer - DataChangeService] [DEBUG] Added subscriber - Sources: DCS_WEBSOCKET, DCS_HTTP, DCS_NETWORKTABLES, DCS_VISIONMODULE, DCS_OTHER, Destinations: DCD_UI
[20:29:33] [2025-01-07 01:29:32] [VisionModule - VisionSourceSettables] [INFO] Setting video mode to FPS: 30 Width: 640 Height: 320 Pixel Format: kYUYV
[20:29:33] [2025-01-07 01:29:32] [VisionModule - VisionSourceSettables] [DEBUG] Failed to set video mode!
[20:29:33] [2025-01-07 01:29:32] [VisionModule - VisionSourceSettables] [WARN] Expected at least one of the following properties to be available: [raw_exposure_absolute, raw_exposure_time_absolute, exposure, raw_Exposure]
[20:29:33] [2025-01-07 01:29:32] [VisionModule - VisionSourceSettables] [WARN] Expected at least one of the following properties to be available: [exposure_auto, auto_exposure]
[20:29:33] [2025-01-07 01:29:32] [VisionModule - VisionSourceSettables] [WARN] Expected at least one of the following properties to be available: [white_balance_temperature, WhiteBalance]
[20:29:33] [2025-01-07 01:29:32] [VisionModule - VisionSourceSettables] [DEBUG] No property image_stabilization for HD_Web_Camera , skipping.
[20:29:33] [2025-01-07 01:29:32] [VisionModule - VisionSourceSettables] [DEBUG] No property power_line_frequency for HD_Web_Camera , skipping.
[20:29:33] [2025-01-07 01:29:32] [VisionModule - VisionSourceSettables] [DEBUG] No property scene_mode for HD_Web_Camera , skipping.
[20:29:33] [2025-01-07 01:29:32] [VisionModule - VisionSourceSettables] [DEBUG] No property exposure_metering_mode for HD_Web_Camera , skipping.
[20:29:33] [2025-01-07 01:29:32] [VisionModule - VisionSourceSettables] [DEBUG] No property exposure_dynamic_framerate for HD_Web_Camera , skipping.
[20:29:33] [2025-01-07 01:29:32] [VisionModule - VisionSourceSettables] [DEBUG] No property focus_auto for HD_Web_Camera , skipping.
[20:29:33] [2025-01-07 01:29:32] [VisionModule - VisionSourceSettables] [DEBUG] No property focus_absolute for HD_Web_Camera , skipping.
[20:29:33] [2025-01-07 01:29:32] [VisionModule - PipelineManager] [DEBUG] Creating AprilTag pipeline
[20:29:33] [2025-01-07 01:29:32] [WebServer - DataChangeService] [DEBUG] Added subscriber - Sources: DCS_WEBSOCKET, DCS_HTTP, DCS_NETWORKTABLES, DCS_VISIONMODULE, DCS_OTHER, Destinations: DCD_ACTIVEMODULE, DCD_ACTIVEPIPELINESETTINGS, DCD_GENSETTINGS, DCD_UI, DCD_WEBSERVER, DCD_OTHER
[20:29:33] [2025-01-07 01:29:32] [VisionModule - VisionModule - HD_Web_Camera] [INFO] Setting pipeline to 0
[20:29:33] [2025-01-07 01:29:32] [VisionModule - VisionModule - HD_Web_Camera] [INFO] Pipeline name: New Pipeline
[20:29:33] [2025-01-07 01:29:32] [VisionModule - VisionSourceSettables] [INFO] Setting video mode to FPS: 30 Width: 640 Height: 320 Pixel Format: kYUYV
[20:29:33] [2025-01-07 01:29:32] [VisionModule - VisionSourceSettables] [DEBUG] Failed to set video mode!
[20:29:33] [2025-01-07 01:29:32] [General - TimedTaskManager] [ERROR] java.lang.NullPointerException: Cannot invoke "edu.wpi.first.cscore.VideoProperty.getName()" because "this.exposureAbsProp" is null at org.photonvision.vision.camera.USBCameras.GenericUSBCameraSettables.setExposureRaw(GenericUSBCameraSettables.java:213) at org.photonvision.vision.processes.VisionModule.setPipeline(VisionModule.java:430) at org.photonvision.vision.processes.VisionModule.<init>(VisionModule.java:160) at org.photonvision.vision.processes.VisionModuleManager.addSources(VisionModuleManager.java:63) at org.photonvision.vision.processes.VisionSourceManager.tryMatchCams(VisionSourceManager.java:121) at org.photonvision.common.util.TimedTaskManager$CaughtScheduledThreadPoolExecutor.lambda$wrap$0(TimedTaskManager.java:56) at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539) at java.base/java.util.concurrent.FutureTask.runAndReset(FutureTask.java:305) at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:305) at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136) at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635) at java.base/java.lang.Thread.run(Thread.java:840)
```

Nothing else in camera troubleshooting seemed to help. Might it be that the camera is a bit too inexpensive and doesn't support some required features (such as setting exposure)?

Sam948-byte, 2025-01-09:
Is the camera showing up in the *Camera Matching* tab, but you're not able to activate it?  Or is it already activated, and the picture isn't good?  If you could also include a screenshot of the *Camera Matching* tab, that would be great.

brunsa2, 2025-01-09:
No evidence of a camera shows up in the web UI. I only see "Placeholder Camera".

Sam948-byte, 2025-01-09:
Are you in the *Camera Matching* tab?  Can you send a screenshot of what you're seeing in that tab specifically?

brunsa2, 2025-01-09:
[image]
[image]
[image]

mcm001, 2025-01-09:
If you list controls with v4l2-ctl --list-ctrls (iirc) what do you see? It might also be worth trying the latest dev

brunsa2, 2025-01-09:
I can likely get a new cam to try before I can get that output. I’ll still get the output, but given the current symptoms, might I want to get that different webcam?

mcm001, 2025-01-10:
You could try a newer version of PhotonVision, which may or may not more gracefully handle weird cameras. but it might be easier to grab a lifecam or something.

brunsa2, 2025-01-10:
Different camera worked. I'm going to return the first cameras.

Any information you would like me to gather from either model?

mcm001, 2025-01-10:
If you know the usb VID/PID/Product string (probably present in your logs), we can add the Completely Broken Quirk to this camera in QuirkyCamera.java I suppose

brunsa2, 2025-01-18:
From `lsusb`: 05a3:9331 ARC International Camera

Marketed as Aluratek HD Webcam, model AWC04F