PhotonVision/photonvision #1704: Relicense `photon-serde` as MIT

gerth2, 2025-01-11:
`serde` (currently GPLv3) is generating source files which are distributed with `photon-lib` (intended to be MIT). This is technically incompatible, as it implies `photon-lib` is actually GPLv3, not MIT as intended. 

While we don't expect any individual FRC team to hit the threshold of `distributing` their software that uses `photon-lib`, we'd like to be technically correct anyway.

Change `serde`'s licensing to be MIT (new license.txt file, new headers) to keep `photon-lib` entirely MIT.

As of commit `e40c8fbca08e8397e00e35f72809ac69d752ad57` on `main`, contributors are:

```txt
 git shortlog -n -s -- photon-serde
     4  Lucien Morey
     3  Matt
     1  Jade
```

@LucienMorey @mcm001 @spacey-sooty before this proceeds, can you comment below with an explicit "agree" below that code you wrote will be relicensed from GPLv3 to MIT ?

mcm001, 2025-01-11:
Yes, I'm happy to release my contributions to photon-serde and the autocode it generates under MIT.  

I think we will still need to relicense photon-targeting right?

spacey-sooty, 2025-01-11:
I am also happy with any of my contributions to photon-serde to be MIT.

LucienMorey, 2025-01-11:
I agree

auscompgeek, 2025-01-11:
I can split out the `photon-serde` change from #1702 if you'd like.

PR #1709 (2025-01-11): photon-serde: Relicense generated files under MIT
Fixes #1704
closed 2025-01-11 by mcm001