PhotonVision/photonvision #1718: Random segfaults in some unit tests

mcm001, 2025-01-13:
**Describe the bug**
A clear and concise description of what the bug is and what the expected behavior should have been.

**To Reproduce**

See: https://github.com/PhotonVision/photonvision/actions/runs/12738416647/job/35500709893?pr=1715 and similar for random setfaults

**Additional context**

Running a photonlibTest from https://github.com/PhotonVision/photonvision/pull/1714 :

```
Failure 1:

[==========] 52 tests from 7 test suites ran. (1505 ms total)
[  PASSED  ] 52 tests.
AddressSanitizer:DEADLYSIGNAL
=================================================================
==77094==ERROR: AddressSanitizer: SEGV on unknown address 0x000069800042 (pc 0x155550999d79 bp 0x1555499ff5a0 sp 0x1555499fed18 T9)
==77094==The signal is caused by a READ memory access.
    #0 0x155550999d79  (/lib/x86_64-linux-gnu/libc.so.6+0x199d79)
    #1 0x155554a942e9 in MemcmpInterceptorCommon(void*, int (*)(void const*, void const*, unsigned long), void const*, void const*, unsigned long) ../../../../src/libsanitizer/sanitizer_common/sanitizer_common_interceptors.inc:881
    #2 0x155554a94bc6 in __interceptor_memcmp ../../../../src/libsanitizer/sanitizer_common/sanitizer_common_interceptors.inc:892
    #3 0x155554a94bc6 in __interceptor_memcmp ../../../../src/libsanitizer/sanitizer_common/sanitizer_common_interceptors.inc:887
    #4 0x15555549ac00 in std::char_traits<char>::compare(char const*, char const*, unsigned long) /usr/include/c++/11/bits/char_traits.h:389
    #5 0x15555549ac00 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::compare(char const*) const /usr/include/c++/11/bits/basic_string.tcc:1444
    #6 0x15555549ac00 in bool std::operator==<char, std::char_traits<char>, std::allocator<char> >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const*) /usr/include/c++/11/bits/basic_string.h:6250
    #7 0x15555549ac00 in GetSinkStreamValues /work/cameraserver/src/main/native/cpp/cameraserver/CameraServer.cpp:163
    #8 0x15555549ac00 in UpdateStreamValues /work/cameraserver/src/main/native/cpp/cameraserver/CameraServer.cpp:228
    #9 0x15555549ac00 in operator() /work/cameraserver/src/main/native/cpp/cameraserver/CameraServer.cpp:464
    #10 0x155554591e9e in std::function<void (cs::RawEvent const&)>::operator()(cs::RawEvent const&) const /usr/include/c++/11/bits/std_function.h:590
    #11 0x155554591e9e in cs::impl::NotifierThread::DoCallback(std::function<void (cs::RawEvent const&)>, cs::RawEvent const&) /work/cscore/src/main/native/cpp/Notifier.h:51
    #12 0x155554591e9e in wpi::CallbackThread<cs::impl::NotifierThread, cs::RawEvent, cs::impl::ListenerData, cs::RawEvent>::Main() /work/wpiutil/src/main/native/include/wpi/CallbackManager.h:165
    #13 0x1555537578da in operator() /work/wpiutil/src/main/native/cpp/SafeThread.cpp:81
    #14 0x1555537578da in __invoke_impl<void, wpi::detail::SafeThreadOwnerBase::Start(std::shared_ptr<wpi::SafeThreadBase>)::<lambda()> > /usr/include/c++/11/bits/invoke.h:61
    #15 0x1555537578da in __invoke<wpi::detail::SafeThreadOwnerBase::Start(std::shared_ptr<wpi::SafeThreadBase>)::<lambda()> > /usr/include/c++/11/bits/invoke.h:96
    #16 0x1555537578da in _M_invoke<0> /usr/include/c++/11/bits/std_thread.h:259
    #17 0x1555537578da in operator() /usr/include/c++/11/bits/std_thread.h:266
    #18 0x1555537578da in _M_run /usr/include/c++/11/bits/std_thread.h:211
    #19 0x1555514dc252  (/lib/x86_64-linux-gnu/libstdc++.so.6+0xdc252)
    #20 0x155550894ac2 in start_thread nptl/pthread_create.c:442
    #21 0x15555092684f  (/lib/x86_64-linux-gnu/libc.so.6+0x12684f)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (/lib/x86_64-linux-gnu/libc.so.6+0x199d79) 
Thread T9 created by T0 here:
    #0 0x155554a58685 in __interceptor_pthread_create ../../../../src/libsanitizer/asan/asan_interceptors.cpp:216
    #1 0x1555514dc328 in std::thread::_M_start_thread(std::unique_ptr<std::thread::_State, std::default_delete<std::thread::_State> >, void (*)()) (/lib/x86_64-linux-gnu/libstdc++.so.6+0xdc328)

==77094==ABORTING

Failure 2:

[==========] 52 tests from 7 test suites ran. (1515 ms total)
[  PASSED  ] 52 tests.
terminate called after throwing an instance of 'std::bad_function_call'
  what():  bad_function_call
./run-photonlibpy-to-crash.sh: line 31: 97288 Aborted                 "$PROGRAM"
(.venv) matt@photonvision:~/Documents/GitHub/photonvision$ 

Failure 3:

[==========] 52 tests from 7 test suites ran. (1529 ms total)
[  PASSED  ] 52 tests.
AddressSanitizer:DEADLYSIGNAL
=================================================================
==74081==ERROR: AddressSanitizer: SEGV on unknown address 0x00006980003b (pc 0x155554a2cc37 bp 0x00006980004b sp 0x1555499fed20 T9)
==74081==The signal is caused by a WRITE memory access.
    #0 0x155554a2cc37 in bool __sanitizer::atomic_compare_exchange_strong<__sanitizer::atomic_uint8_t>(__sanitizer::atomic_uint8_t volatile*, __sanitizer::atomic_uint8_t::Type*, __sanitizer::atomic_uint8_t::Type, __sanitizer::memory_order) ../../../../src/libsanitizer/sanitizer_common/sanitizer_atomic_clang.h:80
    #1 0x155554a2cc37 in __asan::Allocator::AtomicallySetQuarantineFlagIfAllocated(__asan::AsanChunk*, void*, __sanitizer::BufferedStackTrace*) ../../../../src/libsanitizer/asan/asan_allocator.cpp:621
[... 52 lines cut ...]
    #7 0x15555549ac00 in GetSinkStreamValues /work/cameraserver/src/main/native/cpp/cameraserver/CameraServer.cpp:163
    #8 0x15555549ac00 in UpdateStreamValues /work/cameraserver/src/main/native/cpp/cameraserver/CameraServer.cpp:228
    #9 0x15555549ac00 in operator() /work/cameraserver/src/main/native/cpp/cameraserver/CameraServer.cpp:464
    #10 0x155554591e9e in std::function<void (cs::RawEvent const&)>::operator()(cs::RawEvent const&) const /usr/include/c++/11/bits/std_function.h:590
    #11 0x155554591e9e in cs::impl::NotifierThread::DoCallback(std::function<void (cs::RawEvent const&)>, cs::RawEvent const&) /work/cscore/src/main/native/cpp/Notifier.h:51
    #12 0x155554591e9e in wpi::CallbackThread<cs::impl::NotifierThread, cs::RawEvent, cs::impl::ListenerData, cs::RawEvent>::Main() /work/wpiutil/src/main/native/include/wpi/CallbackManager.h:165
    #13 0x1555537578da in operator() /work/wpiutil/src/main/native/cpp/SafeThread.cpp:81
    #14 0x1555537578da in __invoke_impl<void, wpi::detail::SafeThreadOwnerBase::Start(std::shared_ptr<wpi::SafeThreadBase>)::<lambda()> > /usr/include/c++/11/bits/invoke.h:61
    #15 0x1555537578da in __invoke<wpi::detail::SafeThreadOwnerBase::Start(std::shared_ptr<wpi::SafeThreadBase>)::<lambda()> > /usr/include/c++/11/bits/invoke.h:96
    #16 0x1555537578da in _M_invoke<0> /usr/include/c++/11/bits/std_thread.h:259
    #17 0x1555537578da in operator() /usr/include/c++/11/bits/std_thread.h:266
    #18 0x1555537578da in _M_run /usr/include/c++/11/bits/std_thread.h:211
    #19 0x1555514dc252  (/lib/x86_64-linux-gnu/libstdc++.so.6+0xdc252)
    #20 0x155550894ac2 in start_thread nptl/pthread_create.c:442
    #21 0x15555092684f  (/lib/x86_64-linux-gnu/libc.so.6+0x12684f)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (/lib/x86_64-linux-gnu/libc.so.6+0x199d79) 
Thread T9 created by T0 here:
    #0 0x155554a58685 in __interceptor_pthread_create ../../../../src/libsanitizer/asan/asan_interceptors.cpp:216
    #1 0x1555514dc328 in std::thread::_M_start_thread(std::unique_ptr<std::thread::_State, std::default_delete<std::thread::_State> >, void (*)()) (/lib/x86_64-linux-gnu/libstdc++.so.6+0xdc328)

==89459==ABORTING
(.venv) matt@photonvision:~/Documents/GitHub/photonvision$ 
```

Sam948-byte, 2025-01-13:
This error also occurred here, showing that it's not limited to the MacOS Host Build.

https://github.com/PhotonVision/photonvision/actions/runs/12729129088/job/35480552247?pr=1700

mcm001, 2025-01-13:
I think i found something spooky. [here](https://github.com/wpilibsuite/allwpilib/blob/c5f7a2b4ac5213e890a8a0236bb051c44f6e42f9/cameraserver/src/main/native/cpp/cameraserver/CameraServer.cpp#L163), we iterate over `m_addresses`  from a function called by NotifierThread.

The instance could [only come from this magic static](https://github.com/wpilibsuite/allwpilib/blob/c5f7a2b4ac5213e890a8a0236bb051c44f6e42f9/cameraserver/src/main/native/cpp/cameraserver/CameraServer.cpp#L93) here. The callback comes from [m_videoListener here](<https://github.com/wpilibsuite/allwpilib/blob/main/cameraserver/src/main/native/cpp/cameraserver/CameraServer.cpp#L388>). The callback does capture `this`. Not sure about destruction order or how to fix this, but this -seems- like an allwpilib side issue that will need to be patched there.

Sam948-byte, 2025-01-13:
@mcm001 do u want me to make an issue there, or do you want to handle that?

mcm001, 2025-01-13:
(ack whoops!)

samfreund, 2026-01-22:
> if this is still an issue I'm happy to work on this!
> 
> I'm new here, but I really look up to what you guys do here at photonvision and want to help out in any way I can to this community that has done so much for me already in just my first season of frc!

I'm glad you want to help out! This might not be the best first issue, given that it might be an allwpilib issue that'll need fixes upstream. If you want to take a look at some [good first issues](https://github.com/PhotonVision/photonvision/issues?q=sort%3Aupdated-desc+state%3Aopen+label%3A%22good+first+issue%22), those should all be ones you can pick up!