PhotonVision/photonvision #1728: startCalibration not handled correctly when processing settings changes

mdurrani808, 2025-01-15:
**Describe the bug**
When calling processSettingChanges in VisionModuleChangeSubscriber, the switch statement is not matching to the "startCalibration" case and is throwing an error saying that the prop could not be set due to a NoSuchFieldException (which makes sense, because no such field exists). 

**To Reproduce**
Don't have time to reproduce at this time. Logs here: https://discord.com/channels/725836368059826228/725848198794706994/1328911877283119144

**Screenshots / Videos**
See logs above. 

**Platform:**
 - Hardware Platform (ex. Raspberry Pi 4, Windows x64): Orange Pi 5
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch): Unkown
 - PhotonVision Version: v2025.0.0-beta-8
 - Browser (with Version) (Chrome, Edge, Firefox, etc.):
 - Camera(s) Used: 2 innomaker 9281m USB cameras

**Additional context**
Symptom was that whenever the user would start calibration, they would lose their camera feed, I presume this is because of this error being thrown and start calibration never actually happening.

royerboat, 2025-01-15:
For what it's worth, I tried this again on a different Orange Pi 5 with the latest PhotonVision beta 8 image and another innomaker camera and ran into the same issue.

royerboat, 2025-01-20:
I was able to eventually get the cameras to calibrate, which was the original issue that caused this. However, I'm only able to calibrate if I do not select a resolution during calibration. If I select a resolution, the camera disconnects and I cannot complete the calibration.

samfreund, 2025-06-20:
Seems very similar to #1752, albeit with a different camera model although same maker.