PhotonVision/photonvision #1730: Phantom CSI Camera crashing

LucienMorey, 2025-01-15:
**Describe the bug**
If you leave a CSI camera activated, and remove it while powered off photonvision will constantly segfault on any further startup until the device is plugged in again.
**To Reproduce**
Steps to reproduce the behavior:

1. Image PI with dev build `photonlib-v2025.0.0-beta-8-47-gd78f2b86`
2. activate CSI camera
3. Turn off platform
1. Remove CSI camera
4. Turn on platform
5. Check the journal for the `photonvision` service and which it be sad from now on

**Screenshots / Videos**
``` bash
Jan 15 11:07:21 photonvision java[1202]: [2025-01-15 11:07:21] [Camera - VisionSourceManager] [DEBUG] Creating VisionSource for CameraConfiguration [uniqueName=31f0f7f5-db31-4142-b781-8176ea74c64a, matchedCameraInfo=PVCsiCameraInfo[type=ZeroCopyPicam, basename=HQ Camera, path='/base/soc/i2c0mux/i2c@1/imx477@1a', uniquePath='/base/soc/i2c0mux/i2c@1/imx477@1a'], nickname=HQ_Camera, deactivated=true, cameraQuirks=QuirkyCamera [baseName=unicam, displayName=, usbVid=-1, usbPid=-1, quirks={LifeCamControls=false, ArduCamCamera=false, ArduOV9782Controls=false, See3Cam_24CUG=false, AdjustableFocus=false, ArduOV9281Controls=false, AwbRedBlueGain=true, StickyFPS=false, InnoOV9281Controls=false, PsEyeControls=false, ArduOV9782=false, Gain=true, FPSCap100=false, CompletelyBroken=false, ArduOV2311Controls=false}], FOV=70.0]
Jan 15 11:07:21 photonvision java[1202]: [2025-01-15 11:07:21] [Camera - LibcameraGpuSource] [WARN] You have an unknown sensor connected to your Pi over CSI! This is likely a bug. If it is not, then you will have to set your camera FOV differently based on resolution.
Jan 15 11:07:21 photonvision java[1202]: [2025-01-15 11:07:21] [VisionModule - VisionSourceSettables - HQ_Camera] [INFO] Setting video mode to FPS: 90 Width: 320 Height: 240 Pixel Format: kUnknown
Jan 15 11:07:21 photonvision java[1202]: [2025-01-15 11:07:21] [VisionModule - VisionSourceSettables - HQ_Camera] [DEBUG] Creating libcamera
Jan 15 11:07:21 photonvision java[1202]: [2025-01-15 11:07:21] [VisionModule - VisionSourceSettables - HQ_Camera] [ERROR] Couldn't create a zero copy Pi Camera while switching video modes
Jan 15 11:07:21 photonvision java[1202]: [2025-01-15 11:07:21] [VisionModule - VisionSourceSettables - HQ_Camera] [ERROR] Couldn't destroy a zero copy Pi Camera while switching video modes
Jan 15 11:07:21 photonvision java[1202]: [2025-01-15 11:07:21] [VisionModule - VisionSourceSettables - HQ_Camera] [DEBUG] Starting libcamera
Jan 15 11:07:21 photonvision java[1202]: #
Jan 15 11:07:21 photonvision java[1202]: # A fatal error has been detected by the Java Runtime Environment:
Jan 15 11:07:21 photonvision java[1202]: #
Jan 15 11:07:21 photonvision java[1202]: #  SIGSEGV (0xb) at pc=0x0000007f27f79260, pid=1202, tid=1337
Jan 15 11:07:21 photonvision java[1202]: #
Jan 15 11:07:21 photonvision java[1202]: # JRE version: OpenJDK Runtime Environment (17.0.13+11) (build 17.0.13+11-Debian-2deb12u1)
Jan 15 11:07:21 photonvision java[1202]: # Java VM: OpenJDK 64-Bit Server VM (17.0.13+11-Debian-2deb12u1, mixed mode, sharing, tiered, compressed oops, compressed class ptrs, g1 gc, linux-aarch64)
Jan 15 11:07:21 photonvision java[1202]: # Problematic frame:
Jan 15 11:07:21 photonvision java[1202]: # C  [libphotonlibcamera.so+0x9260]  CameraGrabber::streamConfiguration()+0x0
Jan 15 11:07:21 photonvision java[1202]: #
Jan 15 11:07:21 photonvision java[1202]: # No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
Jan 15 11:07:21 photonvision java[1202]: #
Jan 15 11:07:21 photonvision java[1202]: # An error report file with more information is saved as:
Jan 15 11:07:21 photonvision java[1202]: # /opt/photonvision/hs_err_pid1202.log
Jan 15 11:07:21 photonvision java[1202]: #
Jan 15 11:07:21 photonvision java[1202]: # If you would like to submit a bug report, please visit:
Jan 15 11:07:21 photonvision java[1202]: #   https://bugs.debian.org/openjdk-17
Jan 15 11:07:21 photonvision java[1202]: # The crash happened outside the Java Virtual Machine in native code.
Jan 15 11:07:21 photonvision java[1202]: # See problematic frame for where to report the bug.
Jan 15 11:07:21 photonvision java[1202]: #
Jan 15 11:07:21 photonvision systemd[1]: photonvision.service: Main process exited, code=killed, status=6/ABRT
Jan 15 11:07:21 photonvision systemd[1]: photonvision.service: Failed with result 'signal'.
Jan 15 11:07:21 photonvision systemd[1]: photonvision.service: Consumed 14.480s CPU time.
Jan 15 11:07:22 photonvision systemd[1]: photonvision.service: Scheduled restart job, restart counter is at 2.
Jan 15 11:07:22 photonvision systemd[1]: Stopped photonvision.service - Service that runs PhotonVision.
Jan 15 11:07:22 photonvision systemd[1]: photonvision.service: Consumed 14.480s CPU time.
Jan 15 11:07:22 photonvision systemd[1]: Started photonvision.service - Service that runs PhotonVision.
```

**Platform:**
 - Hardware Platform Raspberry Pi 4, Raspberry Pi 5
 - Network Configuration: PC to platform through switch. Both devices are being served a DHCP lease from home router
 - PhotonVision Version: `photonlib-v2025.0.0-beta-8-47-gd78f2b86`
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Firefox Version 134.0 (64-bit)
 - Camera(s) Used: Official Raspi HQ Cam v1.0 2018

**Additional context**

I tried this with any other old USB cam, and I noticed that the device has another state: disconnected. I guess CSI cams can't get there because that port is always active or something?

mcm001, 2025-01-15:
Do you have /opt/photonvision/hs_err_pid1202.log handy?

LucienMorey, 2025-01-15:
I do have log handy [hs_err_pid1202.log](https://github.com/user-attachments/files/18430762/hs_err_pid1202.log)