PhotonVision/photonvision #1733: RC1 / RC.1 version naming issue

tbowers7, 2025-01-15:
**Describe the bug**
There seems to be inconsistency in naming of PhotonLib target files between "rc1" and "rc.1" with the former associated with the `linuxx86_64` images and the latter with the `linuxathena`, `osxuniversal`, and `windowsx86` images.  This inconsistency makes it impossible to make robot code based on PhotonVision's 2025 RC1 release build both for the robot and in GitHub's CI environment.

**To Reproduce**
Follow either path:
1. Use the [photonlib-v2025.1.1-rc1.json](https://github.com/PhotonVision/photonvision/releases/download/v2025.1.1-rc1/photonlib-v2025.1.1-rc1.json) file from the GitHub releases page in VSCode (the `rc1` variant) to try to build a robot project.  This will crash in VSCode with the error:
```
Could not find photontargeting-cpp-v2025.1.1-rc1-linuxathena.zip (org.photonvision:photontargeting-cpp:v2025.1.1-rc1).
     Searched in the following locations:
         https://maven.photonvision.org/repository/internal/org/photonvision/photontargeting-cpp/v2025.1.1-rc1/photontargeting-cpp-v2025.1.1-rc1-linuxathena.zip
```

2. Alternatively, use the `photonlib.json` file passed through the WPILib "Manage Vendor Libraries" tool, using the [URL in the Photon Vision docs](https://docs.photonvision.org/en/latest/docs/programming/photonlib/adding-vendordep.html).  While this pathway builds properly in VSCode, it causes an crash in GitHub's CI tool with the error:
```
Failed to transform photontargeting-cpp-v2025.1.1-rc.1-linuxx86-64.zip (org.photonvision:photontargeting-cpp:v2025.1.1-rc.1) to match attributes {artifactType=gr-directory, org.gradle.status=release}.
      > Could not find photontargeting-cpp-v2025.1.1-rc.1-linuxx86-64.zip (org.photonvision:photontargeting-cpp:v2025.1.1-rc.1).
        Searched in the following locations:
            https://maven.photonvision.org/releases/org/photonvision/photontargeting-cpp/v2025.1.1-rc.1/photontargeting-cpp-v2025.1.1-rc.1-linuxx86-64.zip
```

**Platform:**
 - Hardware Platform: Pathway 1: MacBook Pro, Intel -- Pathway 2: GitHub Runner (Ubuntu22.04.5 LTS)
 - Network Configuration: N/A
 - PhotonVision Version: `v2025.1.1-rc1` / `v2025.1.1-rc.1`
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): N/A
 - Camera(s) Used: N/A

mcm001, 2025-01-15:
Re ran CI

tbowers7, 2025-01-15:
The vendordep JSON file that comes in through the WPILib "Manage Vendor Libraries" tool is now of the `rc1` variant.

While this allows GitHub CI to pass, the VSCode build still crashes on my local machine with the error:
```
Could not find photontargeting-cpp-v2025.1.1-rc1-linuxathena.zip (org.photonvision:photontargeting-cpp:v2025.1.1-rc1).
     Searched in the following locations:
         https://maven.photonvision.org/repository/internal/org/photonvision/photontargeting-cpp/v2025.1.1-rc1/photontargeting-cpp-v2025.1.1-rc1-linuxathena.zip
```

In other words, only failure mode 1 exists now, but it prevent the build / deploy of code.

mcm001, 2025-01-15:
The url you sent, `https://maven.photonvision.org/releases/org/photonvision/photontargeting-cpp/v2025.1.1-rc1/photontargeting-cpp-v2025.1.1-rc1-linuxathena.zip`, is HTTP 200 for me...

tbowers7, 2025-01-15:
I am having the unusual situation of VSCode not being able to reach the file at `https://maven.photonvision.org/repository/internal/org/photonvision/photontargeting-cpp/v2025.1.1-rc1/photontargeting-cpp-v2025.1.1-rc1-linuxathena.zip`, but my browser can access it just fine.

This is even after restarting VSCode and running `>Java: Clean Java Server Workspace` from the VSCode Command Palette.

Since the Photon Vision issue has been resolved, I will close this issue.  Thanks for cleaning up the `rc1` vs `rc.1` issue!

closed 2025-01-15 by tbowers7