PhotonVision/photonvision #1750: Static IP can be outside of comp-legal range

mcm001, 2025-01-27:
IS: static IP address regex simply checks if the IP address looks like an IPv4 address
https://github.com/PhotonVision/photonvision/blob/78b82e3a96d04f79e0b70746dd975c10c8ff1294/photon-client/src/components/settings/NetworkingCard.vue#L33C1-L39C3
https://stackoverflow.com/a/17871737

Should Be: static IP regex checks if on the comp legal range. for 2025, .6-.19 is legal.
[image]

mcm001, 2025-01-27:
regexr.com/8beu1 suggests that `^((25[0-5]|(2[0-4]|1{0,1}[0-9]){0,1}[0-9])\.){3,3}0?(1[0-9]|10)$` works.

TheTripleV, 2025-01-27:
Nit but .220 and up is also technically allowed. https://docs.wpilib.org/en/stable/docs/networking/networking-introduction/ip-configurations.html#on-the-field-dhcp-configuration

mcm001, 2025-01-27:
Should we allow users to set that? Surely .6 to .19 will be enough for anyone.

TheTripleV, 2025-01-27:
No, you're right

mcm001, 2025-01-27:
good to know for later down the road i guess if Redux puts everything on ethercat

TheComputer314, 2025-03-14:
I got bit by this at 2025MOSE, I static'd a OPi to .4 which somehow worked on our home setup but broke on the field. As an idiot who did this, I believe this idiot-proofing should be implemented 😆