PhotonVision/photonvision #1761: YOLOv11 Model Naming Support

Lunar-Academy, 2025-02-03:
From my testing, creating a YOLOv11 model and quantizing/converting to rknn works with PhotonVision's object detection. However, the model must be named as a YOLOv8 model. From looking at the [RKNN toolkit-2](https://github.com/airockchip/rknn-toolkit2/) examples ([RKNN Model Zoo](https://github.com/airockchip/rknn_model_zoo/tree/main/examples)), the way of running the models seem identical. I wasn't able to find any differences in the files ([YOLOv8](https://github.com/airockchip/rknn_model_zoo/blob/main/examples/yolov8/python/yolov8.py) and [YOLOv11](https://github.com/airockchip/rknn_model_zoo/blob/main/examples/yolo11/python/yolo11.py)) used for processing the models. Because of this, PhotonVision's backend for YOLOv8 detection works for YOLOv11. 

[Model I have been testing with](https://github.com/BobcatRobotics/2025-Reefscape/tree/object-detection/src/main/java/frc/robot/Object%20Detection) (Github issues doesn't let me upload the file directly, so it's a link to my team's code.)

Sam948-byte, 2025-02-03:
This is currently in progress, I'm working on integrating this into the JNI right now.

Lunar-Academy, 2025-02-04:
While you're doing that, do you think you would be able to modify the npu core selection? From what I can tell, PV currently relies on autoselecting the npu cores.

[RKNN Object Detector file](https://github.com/PhotonVision/photonvision/blob/ef82328d740e9495ae2ca10ab4c57c75f4a52f4d/photon-core/src/main/java/org/photonvision/jni/RknnObjectDetector.java#L71)

If the -1 on core selection parameter of the create method were be changed to 210, theoretically PV would be able to effectively utilize all npu cores.

Sam948-byte, 2025-02-04:
I believe that's currently addressed in another PR. Not quite sure of the number.

Lunar-Academy, 2025-02-04:
Ah, got it

PR #1736 (2025-02-09): Add YOLO11 Support
[image]
This adds support for YOLO11 models.

Requires [this](https://github.com/PhotonVision/rknn_jni/pull/15) to be merged prior to merging.
NOTE: Until this is merged into the rknn_jni, this PR will fail certain checks as it depends upon added code in the linked PR.

Can be tested using [these files](https://drive.google.com/drive/folders/11AsRhcUlO9VhhNXE_FFedh2DsslDtJlz?usp=sharing).

Closes #1761
closed 2025-02-09 by mcm001