PhotonVision/photonvision #1771: Arducam B332 (OV9281) USB Camera Exposure Settings Not Working

MirrorCY, 2025-02-10:
**Describe the bug**
The camera exposure options are not functioning correctly. When "Auto Exposure" is clicked and then turned off, the issue is temporarily resolved. However, after performing a "Reset Device," the problem reoccurs. After the reset, the exposure is set to the default value from the first startup (around 20?), rather than the manually set exposure value. The expected behavior is that the manually set exposure value should be retained after power cycling and applied when the device is powered on.

**To Reproduce**
Steps to reproduce the behavior:
1. Open the web UI
2. Set the exposure value (camera does not respond to the set value)
3. Enable "Auto Exposure" (camera does not automatically adjust exposure)
4. Disable "Auto Exposure" (camera exposure is correctly set to the manual value)
5. (Optional) Re-enable "Auto Exposure" (auto exposure works fine)
6. Go to settings and click "Reset Device"
7. Wait for the reboot to complete, then check the camera feed — exposure is not correctly set

**Screenshots / Videos**
https://github.com/user-attachments/assets/2f181051-249a-474b-a459-2127ed04dd58
https://wormhole.app/6YB9lZ#RBCfDVPXCaqYhMnaAN3YZg

**Platform:**
 - Hardware Platform (ex. Raspberry Pi 4, Windows x64): Orange Pi 5
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch): OM5P and a standard switch
 - PhotonVision Version: v2025.1.1
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Chrome 133.0.6943.54
 - Camera(s) Used: Arducam B332 (OV9281)

**Additional context**
logs
[photonvision-logs.zip](https://github.com/user-attachments/files/18735151/photonvision-logs.zip)

MirrorCY, 2025-02-10:
BTW, the manual exposure range is also not functioning properly. In the latter half of the range, it remains stuck at a maximum exposure value, and only values below a certain threshold seem to take effect

mcm001, 2025-02-10:
Have you set the Arducam OV9281 quirk in the camera settings tab?

MirrorCY, 2025-02-10:
Thank you for your quick response. Yes, I have set the Arducam OV9281 quirk in the camera settings tab. You can see the settings in the second video I provided

MirrorCY, 2025-02-13:
Enabling auto exposure once before applying manual exposure parameters resolves this issue
[video](https://hydro.ac/paste/fsjz6cnq)

mcm001, 2025-02-14:
Weird behavior, thanks for reporting. My Arducam here locally doesn't seem to do this? I'll try reproducing it more here.

MirrorCY, 2025-02-15:
> After the restart, the exposure is set to the default value from the first startup (around 20?),

I noticed that after restarting the device, it wasn’t set to the default exposure but instead switched to auto exposure (even though auto exposure was turned off). Maybe you could try looking into this approach for troubleshooting. By the way, have you been able to reproduce this issue on your side?

mcm001, 2025-03-15:
What I was hoping to do was cross-reference `watch -cn 1 sudo v4l2-ctl --list-ctrls` against `sudo journalctl -u photonvision.service -f` and see who was lying to me, but can't reproduce yet.

mcm001, 2025-03-15:
Power cycle reproduced it! 

https://github.com/user-attachments/assets/f4c48933-9ea4-4438-869e-f00a86591b7b

mcm001, 2025-03-15:
Gunna try borrowing @MirrorCY 's hack

mcm001, 2025-03-15:
@MirrorCY can you please try PRing the fix you made here upstream, but hidden behind a check for the OV9281 quirk?

https://github.com/frcnextinnovation/photonvision/commit/6cb139c3f049f30f332116cfccaf583572e8e906

mcm001, 2025-03-16:
Can yall try https://github.com/PhotonVision/photonvision/pull/1814 ?

closed 2025-03-16 by mcm001

MirrorCY, 2025-03-16:
Thanks for the follow-up! We just wrapped up our matches at the CNSH regional. We’ll give this a try as soon as we finish some post-event tasks

RockyXRQ, 2025-03-28:
Hi we tried #1814 and unfortunatly it doesn't work, in our [verision](https://github.com/frcnextinnovation/photonvision) we removed the auto-exposure button and forcely set the camera auto-exposure value to false, these finally fix the problem

mcm001, 2025-04-25:
https://discord.com/channels/725836368059826228/725846784131203222/1351401824040652800

Related cursed behavior about AE changing the exposure range

RockyXRQ, 2025-04-30:
I think this explained why we don't have the exposure issue after we removed all the possibility of camera enble AE