PhotonVision/photonvision #1784: Add doc warning for different apriltag maps and rename map to indicate welded

Sam948-byte, 2025-02-16:


prensing, 2025-02-16:
Can both maps be included, and a selector to choose?

Sam948-byte, 2025-02-16:
I believe the current intention is not to do this. It's under consideration, but as this is only an issue for this year, we'd prefer not to add extra functionality that may prove to be unnecessary.

prensing, 2025-02-16:
I am good with your answer, but in case it sways people, there could be another use case. My team probably will end up with a custom tag map for our practice area. If the feature can allow that as another map and switch between all three, that would be great. But I understand about feature creep, so I would not complain if you don't implement it.

mcm001, 2025-02-16:
Yeah I'm of two minds here. I think there's still a lot of sharp edges around using a custom field layout but not sure if yet another crud interface is the right solution

prensing, 2025-02-16:
One other thought. Can the field "name" be published in NT? That way it can be logged and at least verified after the fact.

Sam948-byte, 2025-02-16:
We can definitely publish the field name. For using a custom map, you're just going to have to upload the .json of your map to the coprocessor. We could have a discussion about adding multiple maps and a chooser, but I think in the vast majority of cases you'll only want one map (the official field map), this year being an exception.

PR #1787 (2025-02-19): fix: docs updates for the different apriltag field layouts
update docs, no other stuff
closes #1784