PhotonVision/photonvision #1792: 4+ Second loop overrun when seeing a tag for the first time

DJ-Laser, 2025-02-22:
**Describe the bug**
Using the Advantagekit talonfx swerve example, we added cameras to or robot using photonvision.
Strangely, after turning the robot on the driver would sometimes lose control for 3-5 seconds, but only once per code initialization. When this happened the robot would either no respond to input if it wasn't moving, or it would keep driving and not respond to any input except the disable key.

The logs showed that the vision subsystem was the problem, taking 4+ seconds in the periodic method. I started debugging by commenting out sections of the code. The culprit was [the function that handled getting the camera results](https://github.com/Mechanical-Advantage/AdvantageKit/blob/main/template_projects/sources/vision/src/main/java/frc/robot/subsystems/vision/VisionIOPhotonVision.java#L50) so I commented out all the pose estimation code. That didn't fix it. I noticed that it happened then `photonCamera.getAllUnreadResults()` returned the max amount of results (20) So I tried clearing the array so that it would be empty, which didn't work. I evenually just decided to replace the `photonCamera.getAllUnreadResults()` with an empty list and that fixed it! I modiied the method to do nothing but call `photonCamera.getAllUnreadResults()` and the bug was back! It seems like something in that function is somehow taking 4+ seconds to execute :(

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the advantage kit vision example.
2. Have 2 cameras plugged in and calibrated, set to 3d mode
3. Start the robot so it cant see any tags
4. Move the robot so it sees a tag for the first time
5. Observe a loop overrun where `Vision.periodic()` is 4+ seconds

**Screenshots / Videos**
If applicable, add screenshots to help explain your problem. Additionally, provide journalctl logs and settings zip export. If your issue is regarding the web dashboard, please provide screenshots and the output of the browser console.

**Platform:**
 - Hardware Platform: Ubuntu Server 24.04.2 Minimal
 - Network Configuration: Ethernet to radio aux1 port, Rio to radio rio port
 - PhotonVision Version: v2025.1.1
 - Browser: Chrome 133.0.6943.127 (Official Build) (64-bit)
 - Camera(s) Used: 2x Arducam ov9281

**Additional context**
Only happens once per robot code init.
Running `PhotonCamera.setVersionCheckEnabled(false);` does not help

Gold856, 2025-04-03:
Based on the template, it looks you might be loading the AprilTagFieldLayout right when you see a tag for the first time. I think if you tried a list with one item in it, you'd see the same result. Try loading the AprilTagFieldLayout by accessing VisionConstants on initialization and see if the problem disappears. That will move the load from when you first see the tag, to startup.

DJ-Laser, 2025-04-03:
that seems to have done the trick! glad it wasn't some strange bug in photonvision and just java being silly about class loading :/

closed 2025-04-03 by Gold856