PhotonVision/photonvision #1798: Ability to use WPILib Alert API for Camera Disconnection

Gold872, 2025-03-03:
Similar to https://github.com/wpilibsuite/allwpilib/issues/7840

It would be very useful for pre-match checking (and post-match debugging) to have an alert for whether or not the camera is disconnected. Last season (2024) my team had an alert that updated whenever a camera was disconnected, and it was very useful and even prevented issues before playoff matches.

mcm001, 2025-03-03:
Doesn't seem hard to hook into our current camera stream disconnected logic. Plz pr it :))))))

Gold872, 2025-03-03:
> Plz pr it :))))))

I'll get back to you within 5-7 business days

Also if I PR it a translation to at least one non-java language will be appreciated

Gold872, 2025-03-03:
> camera stream disconnected logic

I'm assuming you're referring to the timesync disconnection warning. My one concern with this is the time on that is 5 seconds, which is quite high. Something like 0.5 seconds (like the heartbeats) is more desirable, but I don't think it makes a ton of sense to use different time sources between `DriverStation.reportWarning()` and the WPILib Alerts

PR #1799 (2025-03-14): Add alerts for timesync and disconnection
Resolves #1798

Also adds missing C++ implementation for `IsConnected()` along with heartbeat checking.

Has not been tested yet.
closed 2025-03-14 by mcm001