PhotonVision/photonvision #1801: Time Sync Server

JDTheBomb, 2025-03-04:
**Describe the bug**
I get warning messages after deploying code to our robot about the cameras time sync server. When this warning happens, photonvision pose estimator stops working.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone YAGSL example code from BroncBotz3481.
2. Edit it for your robot.
3. Push code to robot.
4. Get error "Photon Vision camera is not connected to TimeSyncServer?"
5. Pose estimator not working.
6. Turn robot off
7. Turn robot back on
8. Pose Estimator starts working.
9. Push code again.
10. Get same warning warning.
11. Pose Estimator Stops Working

**Platform:**
 - Hardware Platform (ex. Raspberry Pi 4, Windows x64): Orange Pi 5 Plus
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch): Connected directly to Vivid Hosting router on AUX2
 - PhotonVision Version: 2025.2.1
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Firefox 136.0
 - Camera(s) Used: ELP Camera [https://www.amazon.com/ELP-Shutter-Computer-120degree-Distortion/dp/B0C3C2YVK9?th=1](url)

"""
PhotonVision coprocessor at path /photonvision/flCam2025 is not connected to the TimeSyncServer? It's been 9223372036854.78s since the coprocessor last heard a pong.
Check /photonvision/.timesync/{COPROCESSOR_HOSTNAME} for more information.

PhotonVision coprocessor at path /photonvision/frCam2025 is not connected to the TimeSyncServer? It's been 9223372036854.78s since the coprocessor last heard a pong.
Check /photonvision/.timesync/{COPROCESSOR_HOSTNAME} for more information.
"""

mcm001, 2025-03-05:
>  Check /photonvision/.timesync/{COPROCESSOR_HOSTNAME} for more information.

What do you see here when things are in both the working and not-working configuration?

GregoryLi360, 2025-03-06:
Gonna add on to this. I think the root of the issue is that every time you deploy code, the FGPATimestamp resets to 0 but the PhotonVision timestamps do not. Therefore, when you first power cycle the robot, everything works as normal, but after you deploy code, it breaks. 

In YAGSL, there is this code in `Vision.updateUnreadResults()` which is used for localization: 
```java
      double mostRecentTimestamp = resultsList.isEmpty() ? 0.0 : resultsList.get(0).getTimestampSeconds();
      double currentTimestamp = Microseconds.of(NetworkTablesJNI.now()).in(Seconds);
      double debounceTime = Milliseconds.of(15).in(Seconds);

      for (PhotonPipelineResult result : resultsList) {
        mostRecentTimestamp = Math.max(mostRecentTimestamp, result.getTimestampSeconds());
      }
      
      if ((resultsList.isEmpty() || (currentTimestamp - mostRecentTimestamp >= debounceTime)) &&
          (currentTimestamp - lastReadTimestamp) >= debounceTime) { ... }
```
Because mostRecentTimestamp gets set to the PV timestamp (which is much larger than the FGPA timestamps), the condition `currentTimestamp - mostRecentTimestamp >= debounceTime` will almost always be false.

mcm001, 2025-03-06:
https://github.com/user-attachments/assets/e71858e9-699f-46ad-842f-1c18460aeb6c
https://github.com/user-attachments/assets/4c0a4fe3-4231-4e23-920f-71e03057e510

I was unable to reproduce this behavior here while restarting either my robot code simulation or photonvision at least

JDTheBomb, 2025-03-08:
[image]This is what it shows, but it only shows once and doesn't repeat it.

closed 2025-03-11 by mcm001