PhotonVision/photonvision #1805: PhotonPoseEstimator allows setting primary and fallback strategy the same

sciencewhiz, 2025-03-09:
**Describe the bug**
I helped a team with a crash on the field where they had set both the primary and fallback strategy to CONSTRAINED_SOLVEPNP. For an unknown reason, it tried to fall back (possibly camera was reported as not calibrated) and thus photonlib was stuck in an infinite loop since it fell back to the same strategy that failed initially. It seems like photonlib should not allow setting the fallback strategy to be the same as the primary strategy. 

It's also not clear to me why the multi-tag pnps are not allowed as fallback strategies, but maybe CONSTRAINED_SOLVEPNP shouldn't be allowed also as a fallback strategy. 

 - PhotonVision Version: 2025.2.1

mcm001, 2025-03-09:
Skull

NoahSimon8, 2025-03-11:
> **Describe the bug** I helped a team with a crash on the field where they had set both the primary and fallback strategy to CONSTRAINED_SOLVEPNP. 

Hi, we're actually the team that was involved with this crash. We're really thankful for your help, though I have a slight correction.

We *did not* have the same primary and fallback strategy, but had MULTI_TAG_PNP_ON_COPROCESSOR as primary and CONSTRAINED_SOLVEPNP as the fallback. All of our cameras were calibrated, though one was unplugged. Hope this helps!

closed 2026-01-11 by mcm001