PhotonVision/photonvision #1812: Support for Orange Pi CM5

JayTicku, 2025-03-14:
**Is your feature request related to a problem? Please describe.**
There are many images available for the Orange Pi 5. My teams wants to use a custom vision camera for next season and there are specific functions we want that have resulted in us deciding to use the Orange Pi CM5. Before we purchase one, I want to know if it is supported by PhotonVision or not. 

**Describe the solution you'd like**
If none of the other Orange Pi 5 images support Orange Pi CM5, I would like an image for the Orange Pi CM5.

**Describe alternatives you've considered**
Alternatives would be the other Orange Pi 5 images available, however I am not sure if they would work with the Orange Pi CM5.

mcm001, 2025-03-17:
It's (probably) not as simple as just producing an image -- you'll likely need breakout board specific device tree changes as well. I would suggest you fork Photon and maintain your own images there, and then potentially contribute upstream later?

JayTicku, 2025-03-18:
I can work on that for my team and possibly submit a pull request later. However, could you please point me in the right direction for how to do this? Am I correct in thinking that I am going to have to modify the "photon-targeting/src/main/java/org/photonvision/common/hardware/Platform.java" file and add support for PhotonVision being able to detect the custom hardware? I'm thinking I would add another check to the isRK3588() that checks if the CM5 is being used. Would this logic be correct or am I going in the wrong direction?

mcm001, 2025-03-18:
Probably yeah, but note I (and other contributors) probably won't be able to help you much with debugging since we don't have either a CM5 or your particular hardware.

JayTicku, 2025-03-18:
Ok, thanks. I can use my mentors for help if I need. One last thing, would you be able to give me a quick summary of how the boot folder and config files for the Limelight are created and where they are found. I think I might need to add in some dtoverlays but I can't find it in the repository.

Gold856, 2025-03-18:
Why do you _need_ an Orange Pi CM5? Depending on what you're doing, this could be as easy as just adding the Orange Pi CM5 image to the [image build matrix](https://github.com/PhotonVision/photon-image-modifier/blob/main/.github/workflows/main.yml#L48). But if you're already doing something custom, I'm wondering if your solution will end being compatible with PV in the first place. What are you doing that _requires_ a CM5 that isn't available with a normal Orange Pi 5?

JayTicku, 2025-03-18:
We want to power the Orange Pi from the Rev Mini Power Module that we are using, but the CM5 and the base board that is sold by Orange Pi are the only things that expose the 5v and GND connections. If I am mistaken, please feel free to correct me and help me out.

JayTicku, 2025-03-18:
My bad. I see in developer chat that you guys were talking about using a USB pigtail and I didn't know that that was a thing. I appreciate the help greatly. My team is a very student-led organization and I apologize for my inexperience in the matter of electronics.

Gold856, 2025-03-18:
All good. We just didn't want you to walk down a path that be overly complicated. Glad you found another way!

Gold856, 2025-04-03:
At this time, we'll probably keep image support to the standard boards.

closed 2025-04-03 by Gold856