PhotonVision/photonvision #1845: Server crash on camera calibration export

awordforthat, 2025-03-24:
**Describe the bug**
I'm getting a 500 error in when I try to export my camera calibration settings from the web client. I'm running the server (version 2025.2.1) on an OrangePi B, version. My camera is an Arducam_OV9871_USB. 

**To Reproduce**
Steps to reproduce the behavior:
1. Connect client computer to OrangePi's network
2. Open web client at http://photonvision.local:5800/#/cameras and run through camera calibration
3. Navigate to the specific calibration result and click "Export"
4. Observe the spawned web page showing a black background and the text "Server error". In my case the URL is http://photonvision.local:5800/api/utils/getCalibrationJSON?width=1280&height=720&cameraUniqueName=arducam_ov9281_usb_camera

**Expected behavior**
I expect to either see or be able to download a JSON file containing my camera's calibration settings.

**Screenshots**
This is the browser error I see:

[image]

I looked at journal logs on the Orange Pi and found this null pointer exception:

[image]

**Desktop (please complete the following information):**
I think this is irrelevant, but here it is anyway.
 - OS: Windows 11
 - Project Information: WPILib Information:
Project Version: 2025.3.1
VS Code Version: 1.96.2
WPILib Extension Version: 2025.3.1
C++ Extension Version: 1.23.2
Java Extension Version: 1.38.0
Java Debug Extension Version: 0.58.1
Java Dependencies Extension Version 0.24.1
Java Version: 17
Java Location: C:\Users\Public\wpilib\2025\jdk
Vendor Libraries:
    libgrapplefrc (2025.0.8)
    maplesim (0.3.8)
    PathplannerLib (2025.2.5)
    CTRE-Phoenix (v5) (5.35.1)
    CTRE-Phoenix (v6) (25.3.1)
    photonlib (v2025.2.1)
    ReduxLib (2025.0.1)
    REVLib (2025.0.3)
    Studica (2025.0.1)
    ThriftyLib (2025.0.2)
    URCL (2025.0.1)
    WPILib-New-Commands (1.0.0)
    YAGSL (2025.7.1)

**Additional context**
Add any other context about the problem here.

mcm001, 2025-03-25:
Your logs indicate that you hit `[cameraUniqueName=arducam_ov9281_usb_camera](http://photonvision.local:5800/api/utils/getCalibrationJSON?width=1280&height=720&cameraUniqueName=arducam_ov9281_usb_camera)`. But when I'm in test mode, I download the JSON from `http://localhost:5800/api/utils/getCalibrationJSON?width=1280&height=720&cameraUniqueName=f67df260-074b-4a4b-9df3-08694d488cda`. Note that my cameraUniqueName is a UUID, as I'd expect, while yours is a string. Can you upload your full settings zip (in particular, the database)?

mcm001, 2025-03-25:
https://github.com/Juniormunk/photonvision/commit/1f801cda5a30ed3c25e99199cbea5b08ae640d78#r154266807

[image]

Why are we not preserving case?

closed 2025-03-25 by mcm001