PhotonVision/photonvision #1853: Make strict camera matching more clear to users

samfreund, 2025-03-30:
This should consist of better documentation, telling users that camera matching is done strictly by USB port. Additionally, add info on the camera matching page saying this.

samfreund, 2025-03-30:
@mcm001 can u fix the labels? Front end and docs pls.

KangarooKoala, 2025-03-30:
For additional context- My team had been troubleshooting unrelated issues (which we eventually determined were from a case potentially causing shorts and a questionable Ethernet cable) and had taken our Orange Pi 5 out of the robot. When we put it back into the robot, we didn't remember which cameras went into which ports and didn't realize that the ports mattered. We ended up accidentally plugging the cameras into the wrong ports, causing the cameras to not match up with their previous info:

https://github.com/user-attachments/assets/30c34b50-ccf0-4a67-a7cc-99de7646de75

Because we didn't know that the camera matching was based on the ports, we eventually went on [a XY problem goose chase](https://discord.com/channels/725836368059826228/725848198794706994/1355653896265011332) until we realized that the USB ports did, in fact, matter. If we had known from the start that we should check the USB ports, we could've solved the problem much faster.

PR #1917 (2025-04-20): [docs] Reorganize camera configuration pages and add camera matching documentation
## Description

Reorganizes camera-specific configuration pages for the arducam and pi camera module. This is because the Quick Start pages are becoming cluttered by camera-specific documentation that could be placed elsewhere, allowing for future expansion if more cameras are supported.
Also adds documentation on camera matching.

Resolves #1853 

## Meta

Merge checklist:
- [x] Pull Request title is [short, imperative summary](https://cbea.ms/git-commit/) of proposed changes
- [x] The description documents the [... cut]
closed 2025-04-20 by samfreund