PhotonVision/photonvision #1874: Allow demo to connect to webcam on host device

spacey-sooty, 2025-04-10:
I have no clue how to do this but it'd be sick if you could connect to say a laptop webcam in the demo.

ancientstraits, 2025-04-10:
I feel like this could be possible. You can get cameras from the browser, and WebAssembly also exists, to compile the PHotonVision backend to a format than can be run in browsers. So it is possible, but would take a lot of effort. However, I am interested in WebAssembly, and this seems like it would be a good exercise for me to learn more about using it.

samfreund, 2025-04-10:
@ancientstraits if you're interested in trying it, feel free.  However, this is likely going to be pretty complicated, and it's not necessarily a priority at this time. There's lots of other issues or features that you could also try and work on, but if you really want to do this, we would accept a PR.

spacey-sooty, 2025-04-11:
> I feel like this could be possible. You can get cameras from the browser, and WebAssembly also exists, to compile the PhotonVision backend to a format that can be run in browsers. So it is possible, but would take a lot of effort. However, I am interested in WebAssembly, and this seems like it would be a good exercise for me to learn more about using it.
> 
> I realized that the PhotonVision backend is written partly in C++, and mostly in Java. It would be possible to compile all the C++, and then run the Java with CheerpJ, but making sure everything works well (since PhotonVision depends on a lot of external libraries like OpenCV and WPILib) would be a difficult process.

I was also thinking this could be done with web assembly 🙂 I'd love to see some work on this as I think it'd be really cool. Happy to review and test out any PRs you make

mcm001, 2025-04-11:
I mean atp you’re just re-writing all of photon right?

spacey-sooty, 2025-04-11:
I think you could hypothetically compile it to WASM and ship it in the browser. Not super sure how you'd ship it though.