PhotonVision/photonvision #1895: Support throttling check for OPI

samfreund, 2025-04-14:
https://discord.com/channels/725836368059826228/725848198794706994/1361421542931562597

On RPIs, you can see what's causing throttling here:

[image]

This isn't currently supported on OPIs, but it would be great if it was.

Gold856, 2025-04-17:
We might not be able to list a cause, since I'm not sure Orange Pi has anything that will tell us, but we can check for throttling by checking the `CPU(s) scaling MHz` field in `lscpu`:
No throttling:
```
pi@photonvision:~$ lscpu
Architecture:           aarch64
  CPU op-mode(s):       32-bit, 64-bit
  Byte Order:           Little Endian
CPU(s):                 8
  On-line CPU(s) list:  0-7
Vendor ID:              ARM
  Model name:           -
    Model:              Rockchip RK3588S
    Thread(s) per core: 0
    Core(s) per socket: 0
    Socket(s):          0
    Stepping:           0x2
    BogoMIPS:           48.00
    Flags:              fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimd
                        dp
  Model name:           -
    Model:              0
    Thread(s) per core: 1
    Core(s) per socket: 1
    Socket(s):          1
    CPU(s) scaling MHz: 100%
    CPU max MHz:        1800.0000
    CPU min MHz:        408.0000
    BogoMIPS:           48.00
    Flags:              fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimd
                        dp
  Model name:           -
    Model:              Rockchip RK3588S
    Thread(s) per core: 0
    Core(s) per socket: 0
    Socket(s):          0
    Stepping:           0x2
  Model name:           -
    Model:              0
    Thread(s) per core: 1
    Core(s) per socket: 1
    Socket(s):          1
    CPU(s) scaling MHz: 100%
    CPU max MHz:        1800.0000
    CPU min MHz:        408.0000
    BogoMIPS:           48.00
    Flags:              fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimd
                        dp
  Model name:           -
    Model:              Rockchip RK3588S
    Thread(s) per core: 0
    Core(s) per socket: 0
    Socket(s):          0
    Stepping:           0x2
  Model name:           -
    Model:              0
    Thread(s) per core: 1
    Core(s) per socket: 1
    Socket(s):          1
    CPU(s) scaling MHz: 100%
    CPU max MHz:        1800.0000
    CPU min MHz:        408.0000
    BogoMIPS:           48.00
    Flags:              fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimd
                        dp
  Model name:           -
    Model:              Rockchip RK3588S
    Thread(s) per core: 0
    Core(s) per socket: 0
    Socket(s):          0
    Stepping:           0x2
[... 197 lines cut ...]
    Model:              0
    Thread(s) per core: 1
    Core(s) per socket: 1
    Socket(s):          1
    CPU(s) scaling MHz: 96%
    CPU max MHz:        2304.0000
    CPU min MHz:        408.0000
    BogoMIPS:           48.00
    Flags:              fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimd
                        dp
  Model name:           -
    Model:              Rockchip RK3588S
    Thread(s) per core: 0
    Core(s) per socket: 0
    Socket(s):          0
    Stepping:           0x4
  Model name:           -
    Model:              0
    Thread(s) per core: 1
    Core(s) per socket: 1
    Socket(s):          1
    CPU(s) scaling MHz: 96%
    CPU max MHz:        2304.0000
    CPU min MHz:        408.0000
```

Gold856, 2025-04-17:
Using `lscpu -e=SCALMHZ%` makes the output much easier to parse:
```
SCALMHZ%
    100%
    100%
    100%
    100%
    100%
    100%
    100%
    100%
```

Gold856, 2025-04-17:
Here's the output of `lscpu` on a Raspberry Pi 3b (in case we want to generalize this data):
```
Architecture:             aarch64
  CPU op-mode(s):         32-bit, 64-bit
  Byte Order:             Little Endian
CPU(s):                   4
  On-line CPU(s) list:    0-3
Vendor ID:                ARM
  Model name:             Cortex-A53
    Model:                4
    Thread(s) per core:   1
    Core(s) per cluster:  4
    Socket(s):            -
    Cluster(s):           1
    Stepping:             r0p4
    CPU(s) scaling MHz:   100%
    CPU max MHz:          1200.0000
    CPU min MHz:          600.0000
    BogoMIPS:             38.40
    Flags:                fp asimd evtstrm crc32 cpuid
```
`lscpu -e=SCALMHZ%`:
```
SCALMHZ%
    100%
    100%
    100%
    100%
```

samfreund, 2025-04-24:
x86_64

```
Architecture:             x86_64
  CPU op-mode(s):         32-bit, 64-bit
  Address sizes:          39 bits physical, 48 bits virtual
  Byte Order:             Little Endian
CPU(s):                   32
  On-line CPU(s) list:    0-31
Vendor ID:                GenuineIntel
  Model name:             13th Gen Intel(R) Core(TM) i9-13900HX
    CPU family:           6
    Model:                183
    Thread(s) per core:   2
    Core(s) per socket:   24
    Socket(s):            1
    Stepping:             1
    CPU max MHz:          5400.0000
    CPU min MHz:          800.0000
    BogoMIPS:             4838.40
    Flags:                fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc a
                          rt arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xt
                          pr pdcm sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb ssbd ibrs ibpb stibp ibrs_enhanced
                           tpr_shadow flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid rdseed adx smap clflushopt clwb intel_pt sha_ni xsaveopt xsavec xgetb
                          v1 xsaves split_lock_detect user_shstk avx_vnni dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp hwp_pkg_req hfi vnmi umip pku ospke waitpkg gfni vaes
                           vpclmulqdq rdpid movdiri movdir64b fsrm md_clear serialize arch_lbr ibt flush_l1d arch_capabilities
Virtualization features:  
  Virtualization:         VT-x
Caches (sum of all):      
  L1d:                    896 KiB (24 instances)
  L1i:                    1.3 MiB (24 instances)
  L2:                     32 MiB (12 instances)
  L3:                     36 MiB (1 instance)
NUMA:                     
  NUMA node(s):           1
  NUMA node0 CPU(s):      0-31
Vulnerabilities:          
  Gather data sampling:   Not affected
  Itlb multihit:          Not affected
  L1tf:                   Not affected
  Mds:                    Not affected
  Meltdown:               Not affected
  Mmio stale data:        Not affected
  Reg file data sampling: Mitigation; Clear Register File
  Retbleed:               Not affected
  Spec rstack overflow:   Not affected
  Spec store bypass:      Mitigation; Speculative Store Bypass disabled via prctl
  Spectre v1:             Mitigation; usercopy/swapgs barriers and __user pointer sanitization
  Spectre v2:             Mitigation; Enhanced / Automatic IBRS; IBPB conditional; RSB filling; PBRSB-eIBRS SW sequence; BHI BHI_DIS_S
  Srbds:                  Not affected
  Tsx async abort:        Not affected
```