PhotonVision/photonvision #1918: Stream ports aren't guaranteed

samfreund, 2025-04-18:
The port a stream is opened on is not guaranteed to persist through reboots or deactivations of cameras.

To Repro:

Connect two cameras, observe the ports on the streams.  Deactivate both cameras.  Enable the camera that had the 83-84 ports previously.  Observe how the ports have changed.

Alternatively, observe ports, unplug the camera on 81-82, then reboot.  Observe ports again.

S/B:

Ports don't change.

https://discord.com/channels/725836368059826228/725848198794706994/1362885099938385931

samfreund, 2025-06-29:
This will be implemented by publishing the hostname and camera names for each coprocessor to NT with a key of the coprocessor's MAC address. Then if a conflict is found - either with host names or camera names - the user will be warned to change the conflicting name(s) with a persistent banner.

Gold856, 2025-06-30:
We discussed the possibility of multiple NICs causing multiple MAC addresses to be available, but the consensus is that is an unsupported configuration or only one NIC should be active at a time. Potentially out of scope for this issue, but once we start publishing MAC addresses, we can then publish coprocessor specific info like hardware metrics or time sync data to those topics.

One important piece of functionality we want to keep is PhotonCamera only requiring the name of the camera. Users should not have to worry about which coprocessor a camera is connected to, only the camera name. The warning banners mentioned will be considered sufficient for users to manually fix conflicts should they arise.

PR #1982 (2025-07-04): De-conflict camera names and hostnames by use of a banner
closes #1918 

All this does is warn the user when there's a conflict. Checking is done by publishing the hostname and camera names to NT using the MAC address as a key, then seeing if there are any other coprocs that did that and have the same names.
closed 2025-07-04 by samfreund