PhotonVision/photonvision #1932: Flickering tag causing UI to shift

samfreund, 2025-04-25:
In the web interface, when you have the “Multitag” Estimation enabled, when you are looking at a tag that’s constantly “flickering” as an unstable target, the interface shifts around components as those tag estimates keep showing up and going away. It makes it somewhat impossible to click stuff (this is on the page where you can adjust exposure, brightness, etc.) It’s kind of annoying, and would make adjusting those values a little faster if they didn’t move so much. The reason I bring this up is during field cals we like to change the exposure, and if the tag estimate is flickering, it makes it impossible to drag the slider because the slider position just keeps moving on our screen.

Now, this may also be because our window size was too small or something, but I just wanted to mention it since it did happen so often (every time we did field cals). It’s a really nitpicky quality of life change, but I would request that the tag estimates are “hideable” so you don’t have to deal with it while messing with the settings.

[Originally on CD](https://www.chiefdelphi.com/t/photonvision-2025-feedback/500533/3?u=sam948)

mcm001, 2025-04-29:
MRE: https://github.com/PhotonVision/photonvision/compare/main...mcm001:photonvision:ui-flickering

mcm001, 2025-04-29:
Nothing is jumping out to me as wrong, but to repro, you need to:
- have 2 or 3 settings cards shown side by side
- have the right one be targets, not 3d
- observe that the card's widths change randomly

mcm001, 2025-04-29:
Root cause is either one of these v-rows existing. deleting both makes the flicker go away:
https://github.com/PhotonVision/photonvision/blob/ae9f73130f4c1c2e71e67122a7693bf9cc2f7ed7/photon-client/src/components/dashboard/tabs/TargetsTab.vue#L140-L265

closed 2025-05-06 by Gold856