PhotonVision/photonvision #1937: NetworkTables connecting and disconnecting with PV coprocessor(s)

therekrab, 2025-04-26:
**Describe the bug**
This is not a common issue, but we have experienced issues with PhotonVision and NetworkTables where we see a cycle of NetworkTables connections/disconnections. The issue seems correlated with PhotonVision, because when we put our dashboard, Elastic, next the PV UI, the dashboard disconnects exactly when PhotonVision's UI regains connection. Then, connection is lost on the PV dashboard, but at the same time, Elastic regains communications.

(The use of elastic is not necessary for us to reproduce this bug, but it helped us identify this strange behavior).

We initially encountered the bug when running two coprocessors, each with four cameras attached. Reducing the number of coprocessors to one reduced the rate at which we saw this behavior, but it did not stop it by any means.

The behavior is "fixed" if we restart robot code, which implies that the code we are running may not be the issue. The issue only arises on the first boot of the coprocessor.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn robot and coprocessor on.
2. Sometimes, the bug happens.

**Screenshots / Videos**
Coming soon to a comment near you...
(I'll try to recreate it and screen record driver station, elastic, and the PV UI).

**Platform:**
 - Hardware Platform (ex. Raspberry Pi 4, Windows x64): Orange Pi 5
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch): Network switch. Orange Pi is powered by POE, and is used w/ ethernet to a switch that connects to radio & rio.
 - PhotonVision Version: v2025.3.1
 - Browser (with Version): Doesn't matter; normally don't have one open. We can see the issue by going to the UI via Firefox (v124.0.2).
 - Camera(s) Used: 4 ArduCam OV9281s

**Additional context**
- This is a rare problem, and only occurs once every ~15 power cycles.
- Restarting PV **always** fixes the issue.
- The issue is also persistent across roboRIOs, coprocessors, and cameras. We've made many changes to [our code](https://github.com/hackbots-3414/2025_Reefscape), but nothing seems to work.

Once I get the robot, I will attach logs of the issue occuring (both PV and driverstation).

therekrab, 2025-08-12:
Yes. Both coprocessors (when we used multiple) were set to `10.TE.AM.11` and `10.TE.AM.12`. When we switched to just one coprocessor running four cameras (as opposed to two running eight), we removed the second IP address and left the coprocessor at `10.TE.AM.11`. The programming mentor for our team has said "The deploy and cycle rapidly between connected and disconnected has felt like a resource not being released properly".

Also it's been a while since we've seen this issue, which is strange (it was prevalent up to and including CMP but I don't recall seeing it since). I'll try to do a better job of paying attention and see if I can grab some log files from the next time this happens.

samfreund, 2025-08-15:
Closing as stale for now, but if you see this behavior again don't hesitate to reopen.

closed 2025-08-15 by samfreund