PhotonVision/photonvision #1944: Instability in WPIlib pose estimator after adding vision measurements

Zaque2, 2025-04-30:
**Describe the bug**
Following about 20% of startups, the output of the WPILib pose estimator will slowly spiral out to infinity while receiving vision updates. Resetting the pose estimator almost always results in a return to stability but at least once it has needed to be reset twice. I have not observed the pose estimator become unstable once it is correctly initialized.

**To Reproduce**
I have not been able to determine a correlation between any starting conditions and the issue occurring. We did not have any problems at our first competition where we were purely using odometry; only after the first comp when we started incorporating PhotonVision and added vision measurements to the pose estimator did we start seeing these issues.

**Screenshots / Videos**

**Platform:**
 - Hardware Platform (ex. Raspberry Pi 4, Windows x64): Orange Pi 5
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch): straight to radio
 - PhotonVision Version: 2025.1.1
 - Camera(s) Used: OV 9281

**Additional context**
Add any other context about the problem here.

samfreund, 2025-04-30:
Can you test with 2025.3.1 and see if the issue is still present?

Zaque2, 2025-04-30:
> Can you test with 2025.3.1 and see if the issue is still present?

I did double check the log and it was actually on v2025.2.1, but I will still try to update and run when I am in the shop next (probably Monday).

spacey-sooty, 2025-05-01:
I suspect this is an issue with the WPILib pose estimator as the vision measurements look reasonable and would recommend opening an opening an issue [here](https://https://github.com/wpilibsuite/allwpilib/issues). If you could log your stddevs when reproducing that could also potentially help.

spacey-sooty, 2025-05-12:
Closing after opening an issue against allwpilib

KangarooKoala, 2025-05-12:
@Zaque2 Is this the code link? https://github.com/wildstang/2025_112_robot_software/blob/main/src/main/java/org/wildstang/sample/robot/Robot.java
I'm guessing so based on the log file, but I'd like to make sure.

Zaque2, 2025-05-28:
@KangarooKoala 

Yes, this is the correct repo. We just updated it to reflect the code used at our last regional (where we were experiencing the issue). Let me know if there's anything else I can do to help diagnose.