PhotonVision/photonvision #1986: Disable PV on SystemCore

samfreund, 2025-07-01:
PV takes up a lot of resources, and that's fine when it's running on a coproc. When on the main robot controller, that could cause a lot of unintentional consequences that we don't want to see happen.

samfreund, 2025-07-01:
For bonus points, a little blurb in the quick-start about no support.

mcm001, 2025-07-02:
There are lots of other systemcore-only files we could pull from
- /var/www/html or /var/www/htmlblocks
- if there is a systemcore user
- if there is a systemcore group

samfreund, 2025-07-02:
> There are lots of other systemcore-only files we could pull from
> - /var/www/html or /var/www/htmlblocks
> - if there is a systemcore user
> - if there is a systemcore group

Up to your discretion as to what you think the best approach, I just picked something that I know is easy to check and we know will work.