PhotonVision/photonvision #2002: virus

joe-t-21196, 2025-07-13:
when i tried to open it the anti virus software told me it was infected with a virus

when you open it on command prompt is says its infected

i was on windows and downloaded winx.64

samfreund, 2025-07-13:
While windows defender might identify PhotonVision as a virus, we can assure you that it's not. If you're truly concerned, I suggest building from source so as to ensure that you know where the jar is coming from.

closed 2025-07-13 by samfreund

crschardt, 2025-07-13:
@joe-t-21196 - a couple questions:

1. Which version of the jar did you download?
2. What version of Java is installed on your computer?
3. What command line did you use when running it on the command prompt?
4. Which anti-virus program are you running on your computer?
5. What was the exact message you received from the anti-virus program?