PhotonVision/photonvision #2060: Website missing license

mcm001, 2025-08-12:
https://github.com/PhotonVision/photonvision-website was copied into the monorepo, and had no LICENSE file or copyright headers. Let's round up the troops and get a license added for real

mcm001, 2025-08-12:
@gerth2 @GrahaSH-LLK @mcm001 @spacey-sooty @samfreund @pietroglyph @1337kiwi @Gold856 can we release the website under GPL V3 like the rest of our code?

Gold856, 2025-08-12:
I vote for CC-BY-4.0, but I'm also fine with GPL.

spacey-sooty, 2025-08-13:
I also vote for CC-BY-4.0 but am ok with GPL

gerth2, 2025-08-14:
Especially for "today" - I'm fine with either. I would argue slightly toward CC-BY-4.0 given today's content, it seems to fit better for non-software works.

However, something to consider: the website may at some point host some actual logic (calibration utility? interactive tutorials, a la wpilib's .js controls sims?). At that point, I could make an argument that the website (or at least those code-ish portions) ought to be GPL.

But that's a tomorrow problem, and I have to sleep today.